OpenTabletDriver/OpenTabletDriver #822: Add support for q620m

ColdBytes, 2021-03-02:
# Tablet Information
| Name            | Value |
| --------------- | ----- |
| Manufacturer  | Huion
| Inspiroy Dial Q620M   |
| Width 266.7mm         | 66.7mm 
| Height 166.7mm        | 166.7mm
| Pressure Levels  | 8192

| LPI    5080    |
Report Rate: ＞233PPS
8 Programmable Press Keys + Dial

[diagnostic.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6071802/diagnostic.zip)

Mrcubix, 2021-03-02:
zip fiel appear to be corrupted, re-upload a working one so we can proceed rn

Mrcubix, 2021-03-02:
Go to Programs and features > search for winUSB, attempt uninstall, 

if a windows cover the left side of your screen:
          reinstall huion driver and proceed again, 
else, 
           it will just uninstall, you can then uninstall huion driver

redo a diagnostic once done

Mrcubix, 2021-03-09:
Hi this is a reminder that you opened a ticket for support, i provided all the steps necessary toward getting your tablet closer to working state, i still need informations after proceeding with these steps such as, going again to help > export diagnostic and uploading the new diagnostic file potentially containing new information useful to make this tablet work with OTD.

jamesbt365, 2022-09-12:
Supported now

closed 2022-09-12 by jamesbt365