OpenTabletDriver/OpenTabletDriver #828: Cursor jumping to wrong position

Sudo-kun, 2021-03-06:
## Description

The cursor jumps irregularly from the position my cursor is to position * 5/4 in X direction and position * 3/4 in Y position.
This input jump does not show up on the debug monitor.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System |
| Software Version |
| Tablet           |
OS:                       Manjaro Linux (Kernel 5.10)
Software version: 0.5.1-r164.gec360a1
Tablet:                  Wacom CTL-4100

Sudo-kun, 2021-03-06:
The old wacom drivers still ran in the background, but even after removing xsetwacom it still happend, what do I need to do to only use otd?

X9VoiD, 2021-03-07:
Read [FAQ](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ#i-have-a-ghost-cursor-constantly-alternating-cursor-position)

Sudo-kun, 2021-03-07:
... oh.
Sorry and thanks

closed 2021-03-07 by Sudo-kun