OpenTabletDriver/OpenTabletDriver #830: Unique UX instance broken since 0.5.0

Mrcubix, 2021-03-08:
## Description
You can now open multiple instance of the UX, this can cause an unexpected increase in RAM usage due to known Eto memory leaks.

## System Information:
| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows
| Software Version | 0.5.0 and above

gonX, 2024-11-20:
This has since been fixed in at least 0.6.4.0 from what I can observe. `master` is not running for me but I would imagine it was also fixed there.

gonX, 2025-01-26:
Given this is fixed in 0.6.x I am going to close this. Avalonia currently supports multiple UX instances but it is uncertain to me if this is an important feature to keep. Still, I will note it in the Avalonia megaissue (#3609).

closed 2025-01-26 by gonX