OpenTabletDriver/OpenTabletDriver #837: 0.5.2 Unable to migrate settings from 0.5.0

Mrcubix, 2021-03-09:
## Description

```[Settings:Warning]    Invalid plugin path 'null' has been changed to 'OpenTabletDriver.Desktop.Binding.MouseBinding'```
OTD fail to migrate settings, causing OTD to use default settings for tip button (left click), saving on 0.5.1 and then updating 0.5.2 solve the issue but is only a workaround.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows
| Software Version | 0.5.2

InfinityGhost, 2021-03-09:
This honestly seems like its intentional behavior, don't see how this is a bug.

https://github.com/OpenTabletDriver/OpenTabletDriver/blob/695fd69a3fdeacfb860b953695862ab897c6f9eb/OpenTabletDriver.Desktop/Migration/SettingsMigrator.cs#L15-L16

https://github.com/OpenTabletDriver/OpenTabletDriver/blob/695fd69a3fdeacfb860b953695862ab897c6f9eb/OpenTabletDriver.Desktop/Migration/SettingsMigrator.cs#L59-L68

InfinityGhost, 2021-03-09:
Now I see the issue, null settings are set to defaults when they should be nullable, as in the case of a unbinded button.

PR #843 (2021-03-09): Fix multiple null references and null checks
- Fixed #842
- Fixed #837
  - Default behavior now only defaults it when the path itself is invalid
closed 2021-03-09 by InfinityGhost