OpenTabletDriver/OpenTabletDriver #840: 0.5.2 Seems to break support for Wacom CTL-490 (Needs confirmation)

CaidenBlock, 2021-03-09:
## Description
Updated after playing osu and can no longer get any input from my tablet.
Tablet is detected still.
Debugger shows one report that all auxbuttons are false and then the report rate drops to 0.

## System Information:
| Name             | Value |
| Operating System | Windows 10 20H2 |
| Software Version | 0.5.2 |
| Tablet           | Wacom CTL-490 |

## Notes:
I'm not sure whether this is due to the display changes, as I have a setup with my main monitor and a vertically orientated monitor to the side, but I can't test this case as I only have one tablet.

InfinityGhost, 2021-03-09:
Duplicates of #836 and #835, will be resolved in v0.5.2.1 soon enough. Try out an [artifact](https://github.com/OpenTabletDriver/OpenTabletDriver/actions?query=branch%3Amaster) in the meantime, which should have resolved at least one of these issues.

closed 2021-03-09 by InfinityGhost

CaidenBlock, 2021-03-09:
The latest artifact build does not fix this issue for me, so I'll just stay on 0.5.1 until 0.5.2.1 is released.
:)

CaidenBlock, 2021-03-11:
Not resolved in v0.5.2.1