OpenTabletDriver/OpenTabletDriver #841: No Tablet Dectected

Sausage2580, 2021-03-09:
I have a Gaomon S620 and it can't detect my tablet. I reinstalled it a few times and I'm using the latest installation.

Sausage2580, 2021-03-09:
Wait, can u copy-paste directories for these. I can't really find it.

Sausage2580, 2021-03-09:
I can't find the click update option after right-clicking the device

Mrcubix, 2021-03-09:
in OTD, in OpenTabletDriver.UX.Wpf.exe, go to help > export diagnostic and upload the diagnostic here

Sausage2580, 2021-03-09:
the whole thing?

Sausage2580, 2021-03-09:
{
      "DevicePath": "\\\\?\\hid#vid_256c&pid_006d&mi_01#7&138cbfc7&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 9580,
      "ProductID": 109,
      "CanOpen": false,
      "ReleaseNumberBcd": 256,
      "ReleaseNumber": "1.0",
      "ProductVersion": 256,
      "Manufacturer": "GAOMON",
      "ProductName": "Gaomon Tablet",
      "SerialNumber": "",
      "MaxInputReportLength": 10,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 0
    },
well I think I found the problem

Mrcubix, 2021-03-09:
post the whole thing

Sausage2580, 2021-03-09:
{
  "App Version": "OpenTabletDriver v0.5.2",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.18363.0",
    "VersionString": "Microsoft Windows NT 10.0.18363.0"
  },
  "Environment Variables": {
    "CommonProgramFiles": "C:\\Program Files\\Common Files",
    "ComSpec": "C:\\Windows\\system32\\cmd.exe",
    "OneDrive": "C:\\Users\\isaac\\OneDrive",
    "ProgramW6432": "C:\\Program Files",
    "CommonProgramFiles(x86)": "C:\\Program Files (x86)\\Common Files",
    "SystemRoot": "C:\\Windows",
    "USERPROFILE": "C:\\Users\\isaac",
    "PROCESSOR_IDENTIFIER": "Intel64 Family 6 Model 158 Stepping 13, GenuineIntel",
    "windir": "C:\\Windows",
    "USERDOMAIN_ROAMINGPROFILE": "卢毅升电脑",
    "PSModulePath": "C:\\Program Files\\WindowsPowerShell\\Modules;C:\\Windows\\system32\\WindowsPowerShell\\v1.0\\Modules",
    "TMP": "C:\\Users\\isaac\\AppData\\Local\\Temp",
    "HOMEDRIVE": "C:",
    "SESSIONNAME": "Console",
    "HOMEPATH": "\\Users\\isaac",
    "ProgramFiles": "C:\\Program Files",
    "ProgramFiles(x86)": "C:\\Program Files (x86)",
    "NUMBER_OF_PROCESSORS": "8",
    "COMPUTERNAME": "卢毅升电脑",
    "ProgramData": "C:\\ProgramData",
    "APPDATA": "C:\\Users\\isaac\\AppData\\Roaming",
    "CommonProgramW6432": "C:\\Program Files\\Common Files",
    "PATHEXT": ".COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC",
    "Path": "C:\\Program Files (x86)\\Common Files\\Oracle\\Java\\javapath;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Windows\\System32\\OpenSSH\\;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files\\NVIDIA Corporation\\NVIDIA NvDLISR;C:\\Program Files\\dotnet\\;C:\\Users\\isaac\\AppData\\Local\\Microsoft\\WindowsApps;C:\\Users\\isaac\\AppData\\Local\\GitHubDesktop\\bin;d:\\VSCode\\bin;C:\\Users\\isaac\\AppData\\Local\\atom\\bin",
    "PROCESSOR_REVISION": "9e0d",
    "USERNAME": "isaac",
    "DriverData": "C:\\Windows\\System32\\Drivers\\DriverData",
    "SystemDrive": "C:",
    "TEMP": "C:\\Users\\isaac\\AppData\\Local\\Temp",
    "USERDOMAIN": "卢毅升电脑",
    "LOGONSERVER": "\\\\卢毅升电脑",
    "PROCESSOR_ARCHITECTURE": "AMD64",
    "LOCALAPPDATA": "C:\\Users\\isaac\\AppData\\Local",
    "OneDriveConsumer": "C:\\Users\\isaac\\OneDrive",
    "PROCESSOR_LEVEL": "6",
    "ALLUSERSPROFILE": "C:\\ProgramData",
    "PUBLIC": "C:\\Users\\Public",
    "OS": "Windows_NT"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#hid&col06#1&10a8b2d&1&0005#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 255,
      "ProductID": 47820,
      "CanOpen": true,
      "ReleaseNumberBcd": 1,
      "ReleaseNumber": "0.1",
      "ProductVersion": 1,
      "Manufacturer": "DJP Inc.",
      "ProductName": "Virtual Multitouch Device",
      "SerialNumber": "123123123",
      "MaxInputReportLength": 33,
      "MaxOutputReportLength": 65,
      "MaxFeatureReportLength": 0
    },
    {
      "DevicePath": "\\\\?\\hid#vid_046d&pid_c332&mi_01&col03#7&f7fc48e&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 1133,
      "ProductID": 49970,
      "CanOpen": true,
      "ReleaseNumberBcd": 770,
[... 996 lines cut ...]
      "Time": "2021-03-08T20:09:47.7213813-08:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "Level": 0
    },
    {
      "Time": "2021-03-08T20:09:47.72139-08:00",
      "Group": "IPC",
      "Message": "Disconnected from a client.",
      "Level": 0
    },
    {
      "Time": "2021-03-08T20:09:47.7213987-08:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "Level": 0
    },
    {
      "Time": "2021-03-08T20:09:47.7214074-08:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "Level": 0
    }
  ]
}

Mrcubix, 2021-03-09:
this is the only endpoint, you have a device to update or winUSB to remove

Mrcubix, 2021-03-09:
https://user-images.githubusercontent.com/39861216/110418208-69e58a80-8097-11eb-9079-8abb1adee48c.mp4

from there, right click on gaomon tablet > Click Update -> "Browse my computer" -> "Let me pick" -> HID-compliant device or USB Input Device

Sausage2580, 2021-03-09:
I can't find any of it
[image]

Mrcubix, 2021-03-09:
go to view > device by connection

Sausage2580, 2021-03-09:
I still don't have the click update button
[image]
do I not right-click "Gaomon Tablet"

Sausage2580, 2021-03-09:
oh
ty

Sausage2580, 2021-03-09:
sorry I'm dumb

closed 2021-03-09 by Sausage2580