OpenTabletDriver/OpenTabletDriver #846: Add support for Wacom tablets with wireless adapter.

power9maker, 2021-03-09:
I got it working by copying the CTH-480 config and replacing all `ProductID`'s with 132.
It doesn't add any noticeable latency (at least I can't feel any).

Unfortunately this one cannot be simply fixed by adding more configuration files because the wireless adapter reports itself as `ProductID = 132`, and it doesn't change when using a different tablet, always stays on `132`.
Simplest fix would be adding a window that will pop up whenever the driver detects a device with `VendorID = 1386` (Wacom) and `ProductID = 132`, and the user will have to select what tablet they have.

power9maker, 2021-03-09:
Also, the wireless adapter would have to be set to some kind of lower priority, because right now when you have both tablet and wireless adapter plugged in, OTD selects wireless adapter as the main "tablet", instead of the actual tablet.

power9maker, 2021-03-10:
One more thing is that right now the driver doesn't search for new devices on disconnect, which means when I disconnect the tablet it doesn't register the wireless adapter and I have to manually click `Tablets -> Detect tablet`

adryzz, 2021-03-10:
I think that's because the wireless adapter is still plugged in. 
As far as i understood, the wireless adapter is a HID device itself, not a simple bluetooth adapter with a single bluetooth address enabled (like some mice do). If that's the case then OTD will still recognize it as if the tablet was plugged in.
At this point, i think that the only real way to fix it is a separate menu for wireless tablets.

power9maker, 2021-03-10:
> As far as i understood, the wireless adapter is a HID device itself

Yes, It's a USB dongle.

> If that's the case then OTD will still recognize it as if the tablet was plugged in.

Okay, how do I explain this...
In Windows the adapter is ALWAYS plugged in. If I plug in the tablet with a cable, now Windows sees two "tablets".

> One more thing is that right now the driver doesn't search for new devices on disconnect, which means when I disconnect the tablet it doesn't register the wireless adapter and I have to manually click `Tablets -> Detect tablet`

To demonstrate what I meant here's a gif.
First you can see, the "normal" `CTH-480` is plugged in, but when I disconnect it, it says "no tablet found" and I have to click `Detect tablet` for it to detect the `CTH-480 Wireless` (which is the wireless adapter that's ALWAYS plugged in, and will get detected even if the tablet is turned off).

[image]

adryzz, 2021-03-10:
I think this is expected behaviour when you have multiple tablets plugged in.
I'll look into the code more now, but currently the daemon doesn't check for new tablets when a tablet is disconnected, and since it already was connected, doesn't check back.
I'd say join the discord so we can test this out better.

X9VoiD, 2021-03-10:
OTD -> Help -> Export Diagnostics while the tablet is connected to the wireless dongle. Do the same when tablet is disconnected.

Upload both diagnostics here.

MafinBoy, 2021-03-10:
[Here you go](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6115312/diagnostics.zip)