OpenTabletDriver/OpenTabletDriver #848: Wacom CTL 472 not Being Detected (MAC)

SWAGEDGAR, 2021-03-10:
## Description
Gets detected on wacom driver. I deleted wacom driver still not being detected.

## System Information:
| Operating System | MacOS 10.13.6 High Sierra
| Software Version | 0.5.2.1
| Tablet           | Wacom CTL 472

gionimbus, 2021-03-13:
Same problem with same tablet model on MacOS 10.15.6
[Untitled.json.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6135116/Untitled.json.zip)

makkarakka3, 2021-03-16:
Im having the same issue on Linux Ubuntu 18.04. 2 LTS with the same tablet.

InfinityGhost, 2021-03-16:
> Same problem with same tablet model on MacOS 10.15.6
> [Untitled.json.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6135116/Untitled.json.zip)

Reports as a CTL-672 and its only the auxiliary endpoint, this isn't the same tablet but likely the same issue.

```
    {
      "DevicePath": "IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS02@14200000/CTL-672@14200000/IOUSBHostInterface@1/AppleUserUSBHostHIDDevice",
      "VendorID": 1386,
      "ProductID": 891,
      "ReleaseNumberBcd": 256,
      "CanOpen": true,
      "ReleaseNumber": "1.0",
      "ProductVersion": 256,
      "Manufacturer": "Wacom Co.,Ltd.",
      "ProductName": "CTL-672",
      "SerialNumber": "0LE00M1094328",
      "MaxInputReportLength": 64,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 0
    }
 ```

SWAGEDGAR, 2021-03-22:
@InfinityGhost  I tried clicking on detect tablet console says 
`[Detect:Debug]	Using device 'CTL-472'.
[Detect:Debug]	Using report parser type 'OpenTabletDriver.Tablet.TabletReportParser'.
[Detect:Debug]	Device path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/EHC2@1A,7/EHC2@fa000000/PRT1@fa100000/IOUSBHostDevice@fa100000/AppleUSB20InternalHub@fa100000/AppleUSB20HubPort@fa120000/CTL-472@fa120000/IOUSBHostInterface@0/IOUSBHostHIDDevice@fa120000,0
[IOException:Error]	Failed to open stream
`

InfinityGhost, 2021-03-23:
Really confused how its able to detect the CTL-472 but it didn't show up in diagnostics when you exported. Export diagnostics again, and make sure your tablet is connected for sure.

SWAGEDGAR, 2021-03-23:
> Export diagnostics

I kept unplugging and replugging the tablet and it didn't work for a couple of tries but after it started working. I think it was probably bugged or something.

InfinityGhost, 2021-03-24:
> > Export diagnostics
> 
> I kept unplugging and replugging the tablet and it didn't work for a couple of tries but after it started working. I think it was probably bugged or something.

Potentially an issue in the macOS USB stack, but can't be 100% sure.

InfinityGhost, 2021-07-18:
Stale support request, assuming it has been resolved. This can be reopened in the case this was a mistake.

closed 2021-07-18 by InfinityGhost