OpenTabletDriver/OpenTabletDriver #849: Cursor isn't moving.. CTL-490 / windows 20H2 / portable 0.5.2.1

Fejkej, 2021-03-10:
## Description
I'm on portable version v0.5.2.1. 
Tablet is detected but cursor isn't moving at all even on desktop. Everything is set (tablet area, display area etc.). 

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System |  Windows 20H2 |
| Software Version | 0.5.2.1 |
| Tablet | Wacom CTL-490 |

AbstractQbit, 2021-03-10:
Can you try using this configuration?
[CTL-490.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6116240/CTL-490.zip)

Fejkej, 2021-03-10:
okay.. now its moving but randomly and extrealmy fast... at least tablet debugger shows that it reports something

AbstractQbit, 2021-03-10:
Try this build, should fix everything https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/639869126

Fejkej, 2021-03-10:
thanks! it works now

PR #850 (2021-03-12): Add default device report, fix wacom CTx-x90
Default device report is needed to show all reports in debugger, even if they were not interpreted.

9th gen consumer wacoms seem to report tablet data with 0x10 as the first byte, added that to intuosv2 parser

Closes #849.
closed 2021-03-12 by InfinityGhost