OpenTabletDriver/OpenTabletDriver #851: OpenTabletDriver should not be run with administrator privileges.

ICHINOYA, 2021-03-10:
## Description

I know that a similar question has already been asked, but nothing is clear.
I need a driver with Devocub Filter support.
But when opening this driver, this error constantly appears.
After a while, when playing OSU, the driver simply stops opening (GUI) and has to be restarted.
## Please give a detailed answer!
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10
| Software Version | Latest
| Tablet | GAOMON S620
         
[image]

AbstractQbit, 2021-03-10:
Enable UAC, without it everything you launch runs as admin.

Duplicate of #681

ICHINOYA, 2021-03-10:
> Enable UAC, without it everything you launch runs as admin.
> 
> Duplicate of #681

Thans, but why does it close after a couple of minutes?