OpenTabletDriver/OpenTabletDriver #853: XP-Pen G640 connected and detected, no mouse movement

RobloxUniverse, 2021-03-11:
## Description

I connected my XP-Pen tablet to my pc and fixed a few initial issues already covered on the wiki, but there is no mouse movement when i use my tablet pen, no matter what I do. there are also no errors or the like in the console.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | ElementryOS [based on Ubuntu 18.04.4 LTS] |
| Software Version | v0.5.2.1 |
| Tablet           | XP-Pen Star G640 |

X9VoiD, 2021-03-11:
OTD -> Help -> Export Diagnostics, upload the generated file here

ToxicityCA, 2021-03-11:
[DIAGNOSTIC.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6124877/DIAGNOSTIC.zip)

X9VoiD, 2021-03-11:
There's nothing in the diagnostics. Open OTD -> Console -> Switch log verbosity to debug then copy paste content here as an alternative.

ToxicityCA, 2021-03-11:
The issue was fixed, I rebooted my pc for an unrelated reason, and after opening OTD, the driver did not detect my tablet. Fixed by following steps in the"ArgumentOutOfRangeException: Value [0-15]" of the Linux FAQ:
`echo "blacklist hid_uclogic" | sudo tee -a /etc/modprobe.d/blacklist.conf`
`sudo rmmod hid_uclogic`

closed 2021-03-11 by ToxicityCA