OpenTabletDriver/OpenTabletDriver #854: Support Request:

Narflora, 2021-03-12:
# Support Request
 The driver isn't working anymore
 Im using OpenTabletDriver for like 6 months and it was perfectly fine until the last update. I see the update pop up, i click "update", it updates the driver and it stops working. So the program is fine, it detects the tablet, it even gets input i tested it but my cursor doesn't move nor my clicks don't apply. I tested with hawku driver and my tablet works perfectly fine. I use Wacom CTL-490. Can you please help me i really wanna keep using this driver. 

## What I've already tried

## Diagnostic Information

InfinityGhost, 2021-03-12:
The 490 was bugged, we just fixed this in #850. Try the development build and that should be all you need to get it it up and running.

closed 2021-03-12 by InfinityGhost

Narflora, 2021-03-12:
Thank you so much, its the best driver i’ve ever had tbh. Others are just not compatible with my pc, they often lag or delay. Yours is perfect. Many thanks and keep up the amazing work! Also can you give me advice on learning coding? Im really into it but I don’t know how to or where to start. Anyways, have a great day/night.

InfinityGhost, 2021-03-12:
> can you give me advice on learning coding?

Really the best advice is to just start doing it, even if its pointless stuff. I learned all of what I know through experimentation. Pick a programming language and stick with it.

Narflora, 2021-03-12:
Will do when i have the motivation for it, thanks a lot!