OpenTabletDriver/OpenTabletDriver #860: Linux, cannot save configurations

MichaelVash7886, 2021-03-14:
## Description
Running Manjaro Linux, I was trying to add a tablet config, but everytime I try to save the program just forces closed. Even trying to click "Save configurations to" and selecting the configuration folder crashes the app.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System |  Manjaro
| Software Version |
| Tablet           | GAOMON M10K 2018

Will start a separate issue to get the config added for the tablet.

InfinityGhost, 2021-03-14:
Is this running with the standalone release or from the AUR?

MichaelVash7886, 2021-03-14:
@InfinityGhost The AUR. Glad to try an alternate method if you'd like me to.

InfinityGhost, 2021-03-14:
> @InfinityGhost The AUR. Glad to try an alternate method if you'd like me to.

When you're running from the AUR, OpenTabletDriver is installed to `/usr/share/OpenTabletDriver` which is (typically) write protected from non-root users. The configuration directory is in that same directory, so trying to save there will always fail unless you run as root (which is not recommended). Will add some error message instead of hard crashing like it currently does.

MichaelVash7886, 2021-03-14:
I wondered if that was the issue.
So it sounds like I'd need to CHOWN the `configuration` folder in `/usr/share/OpenTabletDriver` to be owned by my user then it should allow me to save normally. 
I almost wonder if a symlink to somewhere in /home would be better.

PR #1589 (2021-11-01): Add try/catch to prevent crashing when OTD can't save settings/tablet configs
Tiny PR to add an exception catch to Settings.Serialize to handle "no permissions" exception. Prints a log message to console instead, as per InfinityGhost's request [here](https://github.com/OpenTabletDriver/OpenTabletDriver/issues/860#issuecomment-798919593).

Tested via denying myself permission to create/delete files in the config dir (~/.config/OpenTabletDriver), and trying to save settings. Prints log error, doesn't crash, still applies settings.

Fixes #860
closed 2021-11-01 by InfinityGhost