OpenTabletDriver/OpenTabletDriver #861: Add support for GAOMON M10K 2018

MichaelVash7886, 2021-03-14:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | Gaomon
| Product Name    | GAOMON M10K 2018 
| Width           | 10 inches 255
| Height          | 6.25 inches 159mm
| Pressure Levels | Pressure Sensitivity: 8192 levels
| LPI             | 5080 LPI

 Report Rate:233PPS; Sensing Height:10mm
Stats from the Amazon page
[Diagnostic.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6135869/Diagnostic.zip)
[Diagnostic.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6135870/Diagnostic.zip)

## Diagnostic Information

Mrcubix, 2021-03-14:
[M10K.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6135916/M10K.zip)
extract this config to Configuration/Gaomon, then head to the UX > tablets > device string reader > then type the following:
- 9580
- 110
- 201
- send request, copy and paste the obtained string here

MichaelVash7886, 2021-03-14:
@Mrcubix here you go after sending request I got the below: 
OEM02_T17b_190119

Mrcubix, 2021-03-14:
[M10K.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6135945/M10K.zip)

with this config, go to tablet > debugger
move your pen, values will appear, on the left, you can see the reportID, check this value when dragging and when out of range
next, check the max value for position (by positioning your cursor bottom right i assume).

i need all these informations before i can PR the config

MichaelVash7886, 2021-03-14:
Report ID when dragging, 64
Report ID when not touching tablet 0
Position 50800, 31750
And yes you were correct about it being the bottom right.

Mrcubix, 2021-03-14:
[M10K.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6135979/M10K.zip)
this should work properly now

MichaelVash7886, 2021-03-14:
Excellent work, brand new to using a tablet so will have to play around with it to see how it's picking up. Only possible thing that I can tell may be off is that it lists 8 aux buttons when there are 10 aux buttons and it doesn't look like I can do anything with the touch ring on it. May not be able to use that though with this driver.

Thanks for the quick turn around.

Mrcubix, 2021-03-14:
10 aux isn't supported yet and ring isn't either, someone might take a look into it some days

MichaelVash7886, 2021-03-14:
Good to know. Thanks so much for your help.

closed 2021-03-16 by InfinityGhost

beatrice-lettieri, 2025-01-22:
this only works for gaomon m10k, not gaomon m10k 2018, which also has a touch ring

gonX, 2025-01-22:
> this only works for gaomon m10k, not gaomon m10k 2018, which also has a touch ring

See [TABLETS.md](https://github.com/OpenTabletDriver/OpenTabletDriver/blob/d9a4df084357f7cf98e23b18814da4d6502b05da/TABLETS.md) - wheel support is known not supported yet.

PR #862 (2021-03-16): Added Gaomon M10K Configuration
Fixes #861 

Verification:

#861