OpenTabletDriver/OpenTabletDriver #863: Crash caused by setting express key to accented character

GaPala, 2021-03-14:
## Description
Adding an accented character as a key binding to express key in a profile JSON, loading that profile JSON and then pressing that express key results in the driver crashing.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Windows 10 (version 2004) |
| OpenTabletDriver v0.4.2 |
| Gaomon S620  |

Mrcubix, 2021-03-14:
first use the latest version of OTD aka 0.5.2.2, then try again

InfinityGhost, 2021-03-14:
Accented characters aren't supported because it isn't part of the VirtualKey enumerations for any of the platforms, and likely never will be. Try this again on the latest version, if it still occurs (which it likely will) we'll make it silently fail.

GaPala, 2021-03-14:
Confirming that the issue does occur on v0.5.2.2

I personally wouldn't bother with a anti-crash fix, since it's impossible to assign an accent as a key binding with the provided GUI since you need to press ctr+alt+[vowel], but that'd assign the binding to just 'ctrl' anyway (but I guess that's not my choice). The faulty binding is also pretty easy to undo it if you do force it through the JSON, since the driver can still start up and you can change the binding from there as long as you don't use the faulty binding before applying that change.

When support for modifiers in key bindings arrives (#740) that'd basically solve this anyway, since you add accents to vowels like ctrl+alt+e and so just adding ctrl+alt+e as a hotkey will do the same thing as just setting it to output é.

Thanks

closed 2021-03-14 by GaPala

Mrcubix, 2021-03-14:
Try using the script runner plugin and ahk to circumvent that issue