OpenTabletDriver/OpenTabletDriver #864: Settings for Huion Q11K (Non-V2)

seritools, 2021-03-14:
## Description
Hey, I've checked out the configuration for the Q11K (not V2). Since it's in the list as "not tested" I expected to debug myself a bit :)

I had to change a few minor things to get it working:

- My Q11K has the same Product ID as the Q11K V2: I changed `"ProductID": 109,` to `"ProductID": 110,`.
- There seems to be a typo in the device string: It says `"201": "HUION_TI64_\\d{6}$"`, but it should be `"201": "HUION_T164_\\d{6}$"` (note the `I` vs `1`)

After these two changes the device is detected and regular pen operation (including both pen buttons) work fine. I've checked the values in the tablet debugger though, and the `MaxX` and `MaxY` as well as the `Height` should match those of the V2 variant.

The 8 buttons on the tablet itself do not work yet. Pressing them does not change anything in the tablet debugger raw data view either :(

Any advice on how to debug that?

I've added my dump here: [dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6137881/dump.txt)
And my current, modified config here: [Inspiroy Q11K.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6137907/Inspiroy.Q11K.json.txt)

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10 Insider 21322
| Software Version | 0.5.2.2
| Tablet           | Huion Q11K (not V2)

Mrcubix, 2021-03-15:
first of all, don't change setting from an existing variation, copy paste the identifier and you'll obtain something such as
```
{
  "Name": "Huion Inspiroy Q11K",
  "DigitizerIdentifiers": [
    {
      "Width": 279.4,
      "Height": 146.6,
      "MaxX": 51680.0,
      "MaxY": 33020.0,
      "MaxPressure": 8191,
      "ActiveReportID": {
        "Start": 64,
        "StartInclusive": true,
        "End": null,
        "EndInclusive": false
      },
      "VendorID": 9580,
      "ProductID": 109,
      "InputReportLength": 12,
      "OutputReportLength": null,
      "ReportParser": "OpenTabletDriver.Tablet.TabletReportParser",
      "FeatureInitReport": null,
      "OutputInitReport": null,
      "DeviceStrings": {
        "201": "HUION_TI64_\\d{6}$"
      },
      "InitializationStrings": [
        200
      ]
    },
	{
      "Width": 279.4,
      "Height": 174.625,
      "MaxX": 55880.0,
      "MaxY": 34925.0,
      "MaxPressure": 8191,
      "ActiveReportID": {
        "Start": 64,
        "StartInclusive": true,
        "End": null,
        "EndInclusive": false
      },
      "VendorID": 9580,
      "ProductID": 110,
      "InputReportLength": 12,
      "OutputReportLength": null,
      "ReportParser": "OpenTabletDriver.Tablet.TabletReportParser",
      "FeatureInitReport": null,
      "OutputInitReport": null,
      "DeviceStrings": {
        "201": "HUION_T164_\\d{6}$"
      },
      "InitializationStrings": [
        200
      ]
    }
  ],
  "AuxilaryDeviceIdentifiers": [],
  "Attributes": {
    "libinputoverride": "1"
  }
}
```

second, check debugger again, give the reportID for each button presses so we can proceed

seritools, 2021-03-15:
> second, check debugger again, give the reportID for each button presses so we can proceed

Sadly as I wrote, none of the buttons on the tablet seem to work in any way.

Report ID 64: Regular
Report ID 65: Pen button 1
Report ID 66: Pen button 2

None of the buttons on the tablet itself seem to change/send any Report IDs. I've changed ActiveReportID to `(null..null)` in hopes to get all the IDs then, but nope, no change. :(

[image]

Mrcubix, 2021-03-15:
[Inspiroy Q11K.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6143466/Inspiroy.Q11K.zip)

try this config with a different parser and see if there are any differences

seritools, 2021-03-15:
Dang, tried with all parsers, no difference. Then reinstalled the original huion driver and it also doesn't trigger the button bindings o_O Maybe something's broken with my tablet or with the current windows insider version...

Mrcubix, 2021-03-15:
nah i've heard of other tablets with couple aux buttons not working

seritools, 2021-03-15:
Even with the original driver?

InfinityGhost, 2021-03-16:
Honestly this seems like its more of a tablet issue than an OpenTabletDriver issue.

seritools, 2021-03-16:
Yup, I think so as well (quite sad since it mostly just sat in its box :( ).
At least the pen itself works with the config above.

seritools, 2021-03-16:
Oh my god, the tablet has a button lock switch at the side, will test again later!

seritools, 2021-03-16:
Alright, after facepaling a bit and unlocking the buttons, these are the values:

[image]

HW Buttons:
`ReportID: 112`

Fifth byte goes 0x01, 0x02, 0x04, 0x08, 0x10, 0x20, 0x40, 0x80 (`1 << ButtonNumber`) on pressing the buttons, 0x00 on release.

The other bytes are unchanged.

Mrcubix, 2021-03-16:
can you bind them and use them rn?

seritools, 2021-03-16:
Nope, it's detected as cursor position. X: 257, Y: 0x0-0x80

Mrcubix, 2021-03-16:
[Inspiroy Q11K.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6151382/Inspiroy.Q11K.zip)
try this config

seritools, 2021-03-16:
With that the first 4 buttons are detected correctly (but nothing else works now ^^)

Mrcubix, 2021-03-16:
[Inspiroy Q11K.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6151528/Inspiroy.Q11K.zip)
go back to this config, some people may ask you to do some more things bu i don't personally know how to fix your tablet rn

seritools, 2021-03-17:
Maybe @InfinityGhost has an idea now that my noob issues are solved <3

X9VoiD, 2021-03-20:
> 
> 
> [Inspiroy Q11K.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6151382/Inspiroy.Q11K.zip)
> try this config
> 
> as for the position changes when pushing a button, it's due to uc_logic in huion tablets

The configuration here is the correct one, the `GaomonAuxParser` is hardcoded right now to just 4 buttons.

X9VoiD, 2021-03-20:
[Inspiroy Q11K.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6175049/Inspiroy.Q11K.zip)

This should be the correct one, `GaomonReportParser` (to be renamed to `UCLogicParser`) is hardcoded right now to just 4 buttons.

seritools, 2021-03-20:
Can confirm the first 4 buttons now work together with the pen 🎉

closed 2021-05-24 by X9VoiD