OpenTabletDriver/OpenTabletDriver #866: Wacom CTL-472 not being detected on Ubuntu 18.04. 2 LTS

makkarakka3, 2021-03-16:
## Description
Hello! I am trying to get the driver working on Ubuntu 18.04. 2 LTS, but it always says no tablet detected, but i have my tablet plugged in.
## System Information:
| ---------------- | ----- |
| Operating System | Linux Ubuntu
| Software Version | 18.04. 2 LTS
| Tablet           | CTL-472

Mrcubix, 2021-03-16:
use the [latest master action build ](https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/651492004) and see if it's any different

Mrcubix, 2021-03-16:
Check the console for errors message present in the [wiki](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ)

makkarakka3, 2021-03-16:
Did not find in the wiki
Here is the only error i got [DeviceIOException:Error]	Unable to open HID class device (OK).

Mrcubix, 2021-03-16:
[image]
https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ#the-driver-fails-to-open-the-tablet-deviceioexception

makkarakka3, 2021-03-16:
Oh im blind

closed 2021-03-16 by InfinityGhost