OpenTabletDriver/OpenTabletDriver #870: Huion H950P not detected

GeoffreyCoulaud, 2021-03-16:
## Description
Hello, I have a Huion H950P and it is not detected, therefore #761 has a problem somewhere.  
I've also noticed that my tablet's product ID is 110, not 109, maybe there needs to be two different identifiers ?  
I'm willing to dump info about my tablet if anyone can give me a procedure to do so.  

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System| Manjaro Linux |
| Software Version|v0.5.2.2-r0.gdaa463b  | 
| Tablet           | Huion H950P

Mrcubix, 2021-03-16:
you may copy paste the existing identifier and edit the values there, also, you might still encounter errors after that, check the console for the error and the wiki for a fix if it happen

GeoffreyCoulaud, 2021-03-16:
Ok, I just did that, now the GUI console gets this : 
```
[Detect:Info]	Searching for tablet 'Huion H950P'
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
```
And the tablet is not detected in the end.

closed 2021-03-16 by GeoffreyCoulaud