OpenTabletDriver/OpenTabletDriver #881: Tablet stops responding after minimizing the GUI

sant11h, 2021-03-19:
## Description
When I minimize the GUI the tablet stops working, except for the Windows explorer and Chrome (for some reason), but when i alt tab to discord or osu! it just doesn't work anymore. The same situation happens when I only open the Daemon aswell.

## System Information:
| Name             | Value |
| ---------------- | ----- |
| Windows 10 Pro 20H2 - 19042.868 |
| v0.5.2.3 |
| Wacom CTL-490 |

Mrcubix, 2021-03-19:
first don't run OTD as admin, enable UAC to second to top point, and also, define "not working"

sant11h, 2021-03-19:
I'm not running OTD as admin. When I say stop working I mean that the pen doesn't work (i move the pen and nothing happens)

Mrcubix, 2021-03-19:
explorer and you other apps are probably running as admin, enable UAC to second to top point

sant11h, 2021-03-19:
I mainly use the table for osu!, so I opened osu! with no admin privileges and now it works. Personally, I don't want to have UAC enabled, beacuse it's annoying.

Mrcubix, 2021-03-19:
I don't see how clicking on your mouse could be annoying since you're already doing that all the time but you can use the vmulti mode plugin as well as the requirement on it's linked wiki , you'll just have to deal with the added latency of vmulti but we won't be responsible for potential damages due to this added kernel level driver

X9VoiD, 2021-03-19:
OTD is a user-mode driver. It cannot and never will be able to send input to elevated processes due to security reasons enforced by Windows itself. That's why if you disable UAC **all** processes will run *elevated* by default (imagine all malicious code immediately given administrator-level access even without using security vulnerabilities). This is a _**huge security risk**_. And if you still insist on such setup, install [VMultiMode](https://github.com/X9VoiD/VoiDPlugins/wiki/VMultiMode).

Duplicate of #681.

closed 2021-03-19 by InfinityGhost

sant11h, 2021-03-19:
Now I 

> OTD is a user-mode driver. It cannot and never will be able to send input to elevated processes due to security reasons enforced by Windows itself. That's why if you disable UAC **all** processes [...]
Now I understand, thanks for helping.