OpenTabletDriver/OpenTabletDriver #885: Support Request: Huion H950P

owoDeveloper, 2021-03-20:
# Support Request
I know Huion H950P is already in support, but somehow it can't detect my H950P tablet.
The reason why is new version of Huion H950P and older version have diffrent product IDs each other. Mine(device string HUION_T172_180623, older one) has 110, and the newer one has 109.
So, I'll be really appreciated to you if you add support for HUION_T172_180623 tablet.

## What I've already tried
Finding my tablet in Devices list, and then changed H950P's product ID from 109 to 110.

## Diagnostic Information
[h950p_old.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6175002/h950p_old.zip)

X9VoiD, 2021-03-20:
Extract [H950P.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6175027/H950P.zip) to `Configurations\Huion\H950P.json` then press `Detect tablet`.

owoDeveloper, 2021-03-20:
Yes and can you add this file in next release?
I mean, I can't use tablet settings in osu!lazer due to this issue. It's not a big problem but I personally want it to be fixed.

PR #889 (2021-03-20): Update Huion H950P configuration
## Verification

#885

Fixes #885
closed 2021-03-20 by InfinityGhost