OpenTabletDriver/OpenTabletDriver #890: Support XP-Pen Star 06 Wireless

r00ster91, 2021-03-20:
## Description

There is two types of the [XP-Pen Star 06](https://www.xp-pen.com/search/index.html?keyword=06), the Star 06C Wired (that's currently listed in Missing Features) and the Star 06 Wireless which is not currently supported. I would like to see support for it. Perhaps the existing Star 06C Wired code could be used a foundation for the implementation. There is probably many similarities.

X9VoiD, 2021-03-20:
OTD -> Help -> Export Diagnostics, upload the file here.

r00ster91, 2021-03-20:
@X9VoiD what file? Where is Help?

X9VoiD, 2021-03-20:
[image]

X9VoiD, 2021-03-20:
Please do note that we can't start adding support to a new tablet without the diagnostics, even if there is a similar tablet.

r00ster91, 2021-03-20:
@X9VoiD sure, I was asking them if it's possible to export the diagnostics through osu!'s OpenTabletDriver but I didn't get an answer so installed it myself and here it is: [diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6175617/diagnostics.txt)
If it's of any help, here is where I installed the driver manually: https://www.xp-pen.com/download-51.html

X9VoiD, 2021-03-20:
If possible uninstall the other driver, run `sudo rmmod hid_uclogic` then export diagnostics again.

The tablet doesn't follow the expected structure for an XP-Pen tablet. It could be due to it being a wireless one or due to the other driver.

r00ster91, 2021-03-20:
I made sure the driver doesn't run. `sudo rmmod hid_uclogic` gives me `rmmod: ERROR: Module hid_uclogic is not currently loaded`.
[diagnostics2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6175752/diagnostics2.txt)

InfinityGhost, 2021-03-22:
Send the output of `libinput list-devices`, might have to run as root with `sudo` to get the proper output.

r00ster91, 2021-03-22:
`sudo libinput list-devices` doesn't list me any device that looks like a tablet. The amount of listed devices doesn't change regardless of whether it's plugged in or out.

jamesbt365, 2023-08-24:
As there is no comments in over 2 years my best answer is no, not currently. If you have this device, and would like the contribute to supporting it then it would be nice for you to join our [Discord](https://OpenTabletDriver.net/Discord). Alternatively, if you cannot do that, export a diagnostic.