OpenTabletDriver/OpenTabletDriver #892: Better user-facing error messages about common issues

bdach, 2021-03-20:
After integrating OTD in lazer, we are starting to receive issue reports about common issues that are mentioned [in the FAQ](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/General-FAQ) already (see https://github.com/ppy/osu/issues/12101 for an example). For everyone's sanity, I'm thinking it would be good if the errors were described better and if there was a way to guide the users to the destination they need to head to directly, instead of having them post a lazer issue, only then to be directed to an OTD FAQ page.

It would be nice if these common errors were re-exposed in a more user-friendly manner, so that we can tell the user what's wrong exactly and how to fix it. I'm imagining that alongside a human-readable error, an optional URL to the appropriate FAQ section/to the issue creation page could be provided by OTD, so that we may for example redirect the user to said page when they click the error notification. Other applications that might want to integrate OTD may benefit from the same information as well.

InfinityGhost, 2021-03-22:
#893 should resolve this now, still want an outsider POV on this.

bdach, 2021-03-22:
If it's my perspective you're looking for, then I think that error message looks way better now. Thanks!

closed 2021-03-23 by InfinityGhost