OpenTabletDriver/OpenTabletDriver #895: Noticeable tablet smoothing when using CTL-480 on macOS

O1Alweheshy, 2021-03-20:
## Description

The tablet was detected and I was able to set the area just fine. However there is a noticeable degree of smoothing. Not sure if this is only on macOS or if this is something normal with this tablet. 
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | macOS
| Software Version | 11.2.3 
| Tablet           | CTL-480

Mrcubix, 2021-03-20:
Alright let's go through the list real quick:
- OTD driver doesn't have any smoothing filters (nor they are enabled) by default.
- The 480 doesn't have any hardware smoothing.

so my guess is just that something in MacOS is the cause

InfinityGhost, 2021-03-20:
I'm almost certain we're overloading the cursor teleport API causing this smoothing effect, we don't know of any other way to get around this however.

O1Alweheshy, 2021-03-21:
> I'm almost certain we're overloading the cursor teleport API causing this smoothing effect, we don't know of any other way to get around this however.

That's probably to be expected when I am gaming on macOS. However, I remember that I read yesterday that you switched to the original macOS APIs to make it a bit faster. So maybe I should just downgrade to the version directly before that switch and it would fix it? Ah and also do you recommend virtual machines for this problem? Thank you in advance and for the great work you're doing for the community!

InfinityGhost, 2021-03-22:
> However, I remember that I read yesterday that you switched to the original macOS APIs to make it a bit faster. So maybe I should just downgrade to the version directly before that switch and it would fix it? 

I don't remember stating this, I'll have to look into the file history for the macOS pointer handler.

> Ah and also do you recommend virtual machines for this problem?

That's not really necessarily a way to fix, rather a way to work around it. If you feel so inclined to do so, it could be worthwhile as long as you can get decent performance with a macOS hypervisor.

O1Alweheshy, 2021-03-22:
> That's not really necessarily a way to fix, rather a way to work around it. If you feel so inclined to do so, it could be worthwhile as long as you can get decent performance with a macOS hypervisor.

Yeah, you're right it's not really a good solution. I will try to look into the code a bit but I don't have any experience with macOS libraries so it will most probably end up not worthwhile. Thank you again and I wish you a good day!