OpenTabletDriver/OpenTabletDriver #896: GUI crashes immediately upon opening

danielrparks, 2021-03-21:
## Description
I've just built otd from master. Trying to open the gui, it appears for a second but crashes as soon as it connects to the daemon.

otd-gui logs this:
```
Exception in Gtk# callback delegate
  Note: Applications can use GLib.ExceptionManager.UnhandledException to handle the exception.
System.NullReferenceException: Object reference not set to an instance of an object.
   at Gtk.Widget.DrawnSignalCallback(IntPtr inst, IntPtr arg0, IntPtr gch)
   at GLib.ExceptionManager.RaiseUnhandledException(Exception e, Boolean is_terminal)
   at Gtk.Widget.DrawnSignalCallback(IntPtr inst, IntPtr arg0, IntPtr gch)
   at Eto.GtkSharp.Forms.ApplicationHandler.Run()
   at Eto.Forms.Application.Run(Form mainForm)
   at OpenTabletDriver.UX.App.Run(String platform, String[] args) in /home/daniel/.cache/yay/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.UX/App.cs:line 49
   at OpenTabletDriver.UX.Gtk.Program.Main(String[] args) in /home/daniel/.cache/yay/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.UX.Gtk/Program.cs:line 10
```

and otd logs only this:
```
[IPC:Debug]	Connected to a client.
[IPC:Debug]	Disconnected from a client.
```

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Arch Linux |
| Software Version |  ba416c1 |
| Tablet           | n/a |

InfinityGhost, 2021-03-22:
All I can figure is that gtk isn't in your path. Check that you have all of these packages [listed here](https://github.com/OpenTabletDriver/OpenTabletDriver#linux) and report back.

danielrparks, 2021-03-22:
I have `libx11`, `libxrandr`, `libevdev` (provides `libevdev.so.2`), and `gtk3` installed.
Fwiw, here's my gtk3 config: [settings.ini](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6178845/settings.ini.txt)

LavaDesu, 2021-03-22:
I can reproduce this on my arch install as well, likely related to a package update. I have bisected it down to bc77a2d being the first commit to fail.

matiidenys, 2021-03-22:
I got this problem today as well, which is weird, because it worked well yesterday and during this period I didn't update anything on my ArchLinux.
I tried to re-install otd, gtk3, required libs, but nothing yet helped.

danielrparks, 2021-03-22:
Oh, that's interesting! Do you have a backup that you can use to see what changed?

matiidenys, 2021-03-22:
> Oh, that's interesting! Do you have a backup that you can use to see what changed?

Unfortunately, no.

VeryLongNicknameSuchWow, 2021-03-22:
Having the same issue since a few days on Arch Linux using the AUR package `opentabletdriver-git`.

danielrparks, 2021-03-22:
I tried downgrading gtk3 to the version released in February, but I still got the same crash.

VeryLongNicknameSuchWow, 2021-03-22:
I managed to recreate this on an Arch VM. I installed the driver and it worked fine. Afterwards I updated the system and restarted. After the restart the driver was no longer working, so one of the updated packages must have botched it.
Here is the pacman.log https://pastebin.com/cVeZsGtK

danielrparks, 2021-03-22:
Nice! I'll take a look at this later too and see if I can help debug.

InfinityGhost, 2021-03-23:
That makes this less of an OpenTabletDriver issue and more of an Arch Linux issue, but will keep this issue opened to track the progress of when this is resolved.

pakro, 2021-03-23:
I'm not fully convinced this is an Arch Linux issue only. It manifests on Arch Linux probably because of the rolling release. But I'm also skeptical if GTK messed up. I quickly searched both issue trackers and did not find anything related on both bug trackers here https://bugs.archlinux.org/ and https://gitlab.gnome.org/GNOME/gtk/-/issues.

AlexanderS, 2021-03-23:
Just to note it: A downgrade to the previous glib2 (2.66.7-1) fixes this issue.

pakro, 2021-03-24:
Thanks for this information @AlexanderS.
I've created an issue on their bug tracker. 
Here is the link: https://gitlab.gnome.org/GNOME/glib/-/issues/2362

danielrparks, 2021-03-24:
I noticed that Arch [changed the build flags](https://github.com/archlinux/svntogit-packages/commit/e179a799929803f5d5a406127351e7b89f41fba5#diff-37538beb61ff63edebbf735dfcf39e5d732f49183d6beb097169d971875ca422R40) from 2.66.7-1 to 2.68.0-1, so I built glib2 myself with the same build flags as 2.66.7-1, but the issue remained. So it's likely a change in glibc and not the build system.

InfinityGhost, 2021-03-24:
I figure this potentially could be a result of development headers not being included anymore, therefore p/invoke doesn't know where to look. On Ubuntu, you need the `-dev` packages because those headers are only included there. Looks like they remove it with `glib_debug=disabled` but I'm not 100% certain there.

danielrparks, 2021-03-24:
> I figure this potentially could be a result of development headers not being included anymore, therefore p/invoke doesn't know where to look. On Ubuntu, you need the `-dev` packages because those [...]

This is not correct. I'm not sure what it disables (maybe some debugging symbols), but the new version of glib2 in arch contains all the headers of the old one. I also tried compiling and using a current version of glib2 with glib_debug=disabled removed from the command line and it still crashed.

danielrparks, 2021-03-24:
@InfinityGhost What distribution are you using for debugging on Linux? I should be able to make a glib2 package for you that reproduces the crash.

InfinityGhost, 2021-03-24:
> @InfinityGhost What distribution are you using for debugging on Linux? I should be able to make a glib2 package for you that reproduces the crash.

I use Ubuntu 20.10 for testing, but my main distro is NixOS for development prior to testing.

VeryLongNicknameSuchWow, 2021-03-24:
On Ubuntu 20.10 you could temporarily add the `hirsute-proposed` repository and update libglib2.0-0. The version present there is `2.68.0-1`, which does break OTD.
However, be advised that by doing this you would be adding a testing repository of a future Ubuntu release, so more things could potentially break.

danielrparks, 2021-03-24:
On my VM, I temporarily added hirsute-proposed and updated only libglib2.0-0. There was some breakage, but nothing critical, so I would recommend this for testing if you have a VM or an otherwise disposable machine.

LavaDesu, 2021-03-25:
There's an eto issue here picoe/Eto#1914

InfinityGhost, 2021-04-06:
Awaiting an upstream package update in order to resolve this issue, specifically [Eto 2.5.11](https://github.com/picoe/Eto/milestone/27).

LavaDesu, 2021-04-09:
For AUR users, I pushed a temporary stopgap that compiles Eto with a GtkSharp bump, allowing the GUI to work again.

VeryLongNicknameSuchWow, 2021-04-09:
> For AUR users, I pushed a temporary stopgap that compiles Eto with a GtkSharp bump, allowing the GUI to work again.

The build fails for me with the following output https://pastebin.com/3f9Fzc6y

matiidenys, 2021-04-09:
> > For AUR users, I pushed a temporary stopgap that compiles Eto with a GtkSharp bump, allowing the GUI to work again.
> 
> The build fails for me with the following output https://pastebin.com/3f9Fzc6y

The same.

LavaDesu, 2021-04-09:
@VeryLongNicknameSuchWow @matiidenys
That is a separate issue. See pinned comments in the AUR page.

PR #1104 (2021-05-25): Bump Eto.Platform.Gtk to 2.5.11
This fixes GUI not working with glib >= 2.68

Fixes #896 

This may break packaging for maintainers that have already implemented workarounds (AUR maintainer: @LavaDesu, others?)
closed 2021-05-25 by InfinityGhost