OpenTabletDriver/OpenTabletDriver #897: Wacom area

Mrturtlenugget, 2021-03-21:
## Description

The Wacom Area is actually half of what the numbers should be like if i want 100 left i get 50 left
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10
| Software Version | v0.5.2.3
| Tablet           | Any

Mrcubix, 2021-03-21:
Seeing this issue for the first time, i would first assume you did something wrong with the area settings, so first, check if 
 right click > "Lock Aspect Ratio" cause the issue and if it's not the case, make sure to right click > "Lock to usable Area" is disabled when increasing the size. if the problem still persist, then you might be onto something.

InfinityGhost, 2021-03-22:
I need clarification on what this issue is about. Is this about the area offset?

Mrturtlenugget, 2021-03-22:
I didn't do anything but use the Wacom Area becuase if you type the same numbers from let's say whitecat the ratio stays the same but area is halved when converted to mm

InfinityGhost, 2021-03-22:
Then double it? I'm not sure I understand the issue here.

Kuuuube, 2021-03-22:
Wacom calculates area in raw points so if you use a tablet with a higher or lower LPI than the tablet you're copying the area from it will be different. I assume you're attempting to covert a 2540 LPI area to your 5080 LPI tablet which makes the 2540 LPI area half of the physical area since your tablet has physically double the LPI.

Mrturtlenugget, 2021-03-22:
Yeah I used Gaomon S620 and Xp-pen star G640s

InfinityGhost, 2021-03-23:
> Yeah I used Gaomon S620 and Xp-pen star G640s

These tablets have different LPIs, than that of the tablet whitecat uses. This is a non-issue.

closed 2021-03-23 by InfinityGhost