OpenTabletDriver/OpenTabletDriver #898: XP-Pen G430 not detected

Kirass0, 2021-03-21:
## Description

Hi, my tablet is not being detected. It's XP-Pen G430 bought in 2016.
[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6176894/diagnostics.txt)

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10
| Software Version | 0.5.2.3
| Tablet           | XP-Pen G430

X9VoiD, 2021-03-21:
Do you have another driver installed? This G430 is missing a HID endpoint.

Kirass0, 2021-03-21:
Currently I have original drivers installed, but even without them OTD wasn't detecting it.
Should I uninstall and then export diagnostics again?

X9VoiD, 2021-03-21:
Check if you have an installed program called "WinUSB", uninstall that if it's there, and then re-export diagnostics.

Kirass0, 2021-03-21:
Driver uninstalled. I didn't find anything like "WinUSB" in programs list.
[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6176914/diagnostics.txt)

X9VoiD, 2021-03-21:
Unfortunately this G430 don't have an explicit vendor endpoint to communicate with. It cannot be supported on Windows reliably as Windows itself exclusively opens the handle to the device.

InfinityGhost, 2021-03-22:
> Unfortunately this G430 don't have an explicit vendor endpoint to communicate with. It cannot be supported on Windows reliably as Windows itself exclusively opens the handle to the device, leaving us with no way to access it.

Pretty sure this is also a duplicate of a previous attempt to get a G430 working with this issue.

PR #1425 (2021-11-04): Update XP-Pen G430 Configuration
Verification:

https://discord.com/channels/615607687467761684/885624596567560263/886026557636870164

# Note:

- Added WinUSB endpoint as old variants need WinUSB

# Changes:

- Fixes #898 

# Tablet Owner / Contributor:

- wizer#4545
closed 2021-03-23 by InfinityGhost