OpenTabletDriver/OpenTabletDriver #900: Need to rescan for tablet after logging in from locked machine

tianshiz, 2021-03-22:
## Description

Everything works fine at first load of OpenTabletDriver. Then, I lock the machine with ctrl+L, and try to unlock the machine. On unlock, the tablet stops working.

 I need to manually re-open opentabletdriver and "Tablets->detec tables" to get the device working.
 Is this as intended? Is it possible to get the default driver settings applied to work on startup and not die from locking?
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Ubuntu 20.04
| Software Version |  0.5.2.3
| Tablet           | XP-PEN G640

InfinityGhost, 2021-03-23:
There's really no good fix for this kind of issue on Linux due to the vast amount of display managers. Assuming you're running Gnome, its probably gdm which I use myself and can't reproduce your issue.

InfinityGhost, 2021-07-18:
Stale support request, assuming it has been resolved. This can be reopened in the case this was a mistake.

closed 2021-07-18 by InfinityGhost