OpenTabletDriver/OpenTabletDriver #901: Add support for Huion GT-220 V2

tamanor, 2021-03-22:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | Huion
| Product Name    | Huion GT-220 V2
| Width           | 1920
| Height          | 1080
| Pressure Levels | 8192
| LPI             | 5080

## Diagnostic Information

[Diagnostic.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6185745/Diagnostic.zip)

InfinityGhost, 2021-03-23:
```
{
  "DevicePath": "\\\\?\\hid#vid_256c&pid_006e&mi_01#9&14d270d9&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
  "VendorID": 9580,
  "ProductID": 110,
  "CanOpen": false,
  "ReleaseNumberBcd": 0,
  "ReleaseNumber": "0.0",
  "ProductVersion": 0,
  "Manufacturer": "",
  "ProductName": "Tablet Monitor",
  "SerialNumber": "",
  "MaxInputReportLength": 8,
  "MaxOutputReportLength": 0,
  "MaxFeatureReportLength": 0
}
```

jamesbt365, 2021-12-31:
We cannot support this without a string dump, if anyone owns this tablet can they export a diagnostic and input into tablet > device string reader 9580 into the top box 110 in the middle box and then click dump all and send that here too.

WobbleFox, 2021-12-31:
I hope I've done this properly.
Thank you for your attention to this tablet.

[Huion GT-220 V2_Diagnostic and Dump.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7796220/Huion.GT-220.V2_Diagnostic.and.Dump.zip)

jamesbt365, 2021-12-31:
Extract this to Configurations\Huion, then go to tablet > detect tablet, report back here once you have done that.

[gt220v2.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7796244/gt220v2.zip)

WobbleFox, 2021-12-31:
I have done so and the tablet is now detected!

jamesbt365, 2021-12-31:
Go to tablet > tablet debugger, then move you pen to the bottom right of you tablet, get the max x and y, the key to getting the highest value is to move slowly.

WobbleFox, 2021-12-31:
[29800, 53629]

jamesbt365, 2021-12-31:
And the max pressure when pushing down

WobbleFox, 2021-12-31:
8191

jamesbt365, 2021-12-31:
the 29800 is the y axis right?

WobbleFox, 2021-12-31:
It is the number that changes when I move horizontally, and the one listed first in the software.
[image]

jamesbt365, 2021-12-31:
[gt220v2.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7796270/gt220v2.zip)
Replace that config with this one, get the value again

WobbleFox, 2021-12-31:
Instead of 29800, it is now 95336

jamesbt365, 2021-12-31:
what about the other axis?

WobbleFox, 2021-12-31:
It has remained the same at 53629

jamesbt365, 2021-12-31:
Pressure still the same?

WobbleFox, 2021-12-31:
Yes, 8191

jamesbt365, 2021-12-31:
replace config again, [gt220v2.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7796287/gt220v2.zip)
Can you reach the edges of the screen correctly now?

WobbleFox, 2021-12-31:
The cursor travels beyond the bottom of the screen once I travel a bit more than half way down it with my pen.

jamesbt365, 2021-12-31:
Traveling past by half the cursor is normal due to the way windows handles the cursor sprite, am i correct this tablet does not have buttons on the tablet and it has 2 on the pen?

WobbleFox, 2021-12-31:
That's correct.

jamesbt365, 2021-12-31:
This Configuration should be complete unless something else isn't working

jamesbt365, 2021-12-31:
Actually, i may of made a mistake.

WobbleFox, 2021-12-31:
That's much better thank you. The reaches screen edges perfectly now.
Pressure sensitivity does not seem to be working however.

jamesbt365, 2021-12-31:
Do both these things and set your drawing application to windows ink:
https://github.com/X9VoiD/vmulti-bin/releases/tag/v1.0
https://github.com/X9VoiD/VoiDPlugins/wiki/WindowsInk

WobbleFox, 2021-12-31:
Yes! Pen pressure is working now!
Thank you very much.

closed 2021-12-31 by jamesbt365