OpenTabletDriver/OpenTabletDriver #902: Question about OTD with OpenRC.

matiidenys, 2021-03-23:
The question is can this driver be enabled on ArchLinux based distro(Artix Linux), where is OpenRC instead of systemd?
I ask it, because there are steps to enable daemon via systemd in the installation guide, but yet nothing about OpenRC. (I'm not very familiar with OpenRC, but wanted to try without losing ability to use OTD)

X9VoiD, 2021-03-23:
You may always just use the binaries (`OpenTabletDriver.UX.Gtk` and `OpenTabletDriver.Daemon`) directly and make a service file or w/e the equivalent is in OpenRC if you want to automate spawning otd daemon at boot

InfinityGhost, 2021-05-25:
We don't provide service files for neither systemd nor OpenRC, so this is out of scope of this repository.

closed 2021-05-25 by InfinityGhost