OpenTabletDriver/OpenTabletDriver #906: Support Request: OpenTabletDriver can't connect to Daemon & doesn't detect tablet. I have an XP-Pen Star G640, which is supported and has worked in the past (I have used this driver for several months now). But it suddenly stopped working yesterday (NOTE: I use this driver for the game osu!). I have already tried installing

Versedd, 2021-03-23:
# Support Request

- OpenTabletDriver can't connect to Daemon & doesn't detect tablet. >

- I have an XP-Pen Star G640, which is supported and has worked in the past (I have used this driver for several months now). But it suddenly stopped working yesterday (NOTE: I use this driver for the game osu!). [-->](https://i.gyazo.com/876a5d4e68c7b6b1d6295b67e2a5a43c.png)

## What I've already tried
- Uninstalling and reinstalling the driver.
- Unplugging and plugging back in my tablet.
- Installing the .NET Framework runtime (already had it installed).
- Deleting the PenTablet HID in Device Manager

## Diagnostic Information
- Attempting to export the diagnostics just results in another window of the installer popping up. Not sure what to do here.

Mrcubix, 2021-03-23:
use portable and export a diagnostic from there, name it, save it, upload it

InfinityGhost, 2021-04-06:
Almost 2 weeks without response, assuming resolved.

closed 2021-04-06 by InfinityGhost