OpenTabletDriver/OpenTabletDriver #908: Add support for x32 bit systems

eaull, 2021-03-24:
I use windows 32 bit and I can't open the programs in the latest release.

jjhhgg100123, 2021-03-24:
Copied from your previous issue you closed:

OpenTabletDriver doesn't officially support 32 bit operating systems. If you require it, you can build it yourself from the source, shouldn't be too difficult. But support will not be provided for it, as x86 is at this point, ancient, and providing support for it just wouldn't be worth the time it would take.

InfinityGhost, 2021-03-24:
Duplicate of #280

See [here](https://github.com/OpenTabletDriver/OpenTabletDriver/issues/280#issuecomment-690862950) for instructions to compile it from source on Windows x86. There are no guarantees that it will work and no effort will be put in to make it work either if it doesn't.

closed 2021-03-24 by InfinityGhost