OpenTabletDriver/OpenTabletDriver #909: CTL-4100 Buttons always pressed after receiving positional input

GopoBoi, 2021-03-24:
## Description

with the tablet buttons bound to anything, simply placing the pen above the tablet will cause the buttons to be pressed until they're actually physically pressed.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Arch Linux (happens on windows too though) |
| Software Version | v0.5.2.3  |
| Tablet           | Wacom CTL-4100 |

X9VoiD, 2021-03-24:
OTD -> Help -> Export Diagnostics, upload the json file here. Will double check configuration for anything weird.

GopoBoi, 2021-03-24:
can't upload json files directly, it's in the zip file
[diagnostics.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6195902/diagnostics.zip)

X9VoiD, 2021-03-24:
Replace `Configurations\CTL-4100.json` with this [CTL-4100 config](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6196082/CTL-4100.zip) then press `Detect tablet` to test.

GopoBoi, 2021-03-24:
Can confirm that it fixes it.

PR #910 (2021-04-06): Match digitizer/aux parser and remove wild configs
## Changes

- Match the parser used in aux with the digitizer's
- Remove "wild" configs that identifies as InputReportLength: null, OutputReportLength: null. These configs match anything and causes overlaps

Fixes #909
closed 2021-04-06 by InfinityGhost