OpenTabletDriver/OpenTabletDriver #911: Huion H640P is not working

GzFabi, 2021-03-24:
## Description

huion H640P is not working
## System Information:

| Name      | Value | 
| Fabian usernam: GzFabi | ? |
| Operating System |
AMD Athon X2, AMD Radeon HD 6670  
| Software Version |
windows 10 64 bit
| Tablet           |
Huion H640P

X9VoiD, 2021-03-24:
OTD -> Help -> Export Diagnostics, upload the resulting diagnostics here as zip file.

InfinityGhost, 2021-04-06:
Nearly 2 weeks without response, assuming resolved.

closed 2021-04-06 by InfinityGhost

ducanh2002123, 2021-04-14:
Status report: I am using a huion H640P, the software can detect it just fine, but the only action that i could do on the tablet is to tap the pen on the tablet and that would simulate a left click, mouse movement doesn't work, custom button mapping doesn't work, basically only 2 things work: 1: Detecting the tablet, and 2: clicking on the tablet, the console doesn't return any error code

Diagnostic file:
[opentabletdriverdiagnostic.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6312143/opentabletdriverdiagnostic.txt)

X9VoiD, 2021-04-14:
Yoy might want to set an actual tablet area before complaining that the input is not working.

ducanh2002123, 2021-04-15:
Oh..? On the GUI it said my width is 160 and my height is 100, or X:80 and Y:50.
[image]

GzFabi, 2021-04-15:
> Nearly 2 weeks without response, assuming resolved.

> Nearly 2 weeks without response, assuming resolved.

nope

Mrcubix, 2021-04-15:
> OTD -> Help -> Export Diagnostics, upload the resulting diagnostics here as zip file.
> 
> H640P is already supported, just not tested. If you have problems with detection follow instructions from [FAQ](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Windows-FAQ#my-gaomonhuion-tablet-isnt-detected-but-is-listed-as-supported)