OpenTabletDriver/OpenTabletDriver #913: Scrolling by dragging the pen in osu! song selection is dreadful

MilkyIQ, 2021-03-25:
## Description

If I scroll at the speed that I usually do, the song selection will move a little bit and then go back. If I do this process _slowly_, it's fine, but doing it quickly, I'll get nowhere.

From what I can tell it seems to be because the driver is still registering my pen as being pressed down for a brief period after lifting the pen. I was hot-swapping (for lack of a better term) between hawku, Devocub, and OTD drivers, and this one was the only one I had this specific issue with.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10 |
| Software Version | v0.5.2.3 |
| Tablet           | Wacom CTL-490 |

InfinityGhost, 2021-03-25:
Seems like an issue with bindings not releasing on time. What output mode are you using?

MilkyIQ, 2021-03-28:
I'm not exactly sure what you're referring to by output mode. (If you're talking absolute vs relative im running in absolute)

Other than that I didn't change anything in the bindings

MilkyIQ, 2021-03-28:
After messing around with tip pressure, it works just fine now. Turns out it was a binding issue

closed 2021-03-28 by MilkyIQ

AbstractQbit, 2021-07-01:
That's a hardware issue of CTx-x90 tablets, the cursor on those tablets teleports when transitioning between hover and drag.