OpenTabletDriver/OpenTabletDriver #916: Add support for Huion GC610

iceCreamMint, 2021-03-25:
i have a sneaking suspicion that this is just the g10t but sold in china
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | Huion
| Product Name    | GC610
| Width           | 374 mm(254 working)
| Height          | 177 mm(158.8 working)
| Pressure Levels | 8192 levels
| LPI             | 5080

## Diagnostic Information

[gc610.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6204263/gc610.txt)

Mrcubix, 2021-03-25:
[GC610.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6204421/GC610.zip)
with htis config, head to tablet > device string reader, there type the following:
-9580
-110
-201
send request and paste the obtained string here

iceCreamMint, 2021-03-25:
HUION_T166_170329

iceCreamMint, 2021-03-25:
HUION_T166_170329

Mrcubix, 2021-03-25:
[GC610.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6206256/GC610.zip)
try this config, it should work now

iceCreamMint, 2021-03-25:
it works

Mrcubix, 2021-03-25:
alright, i'll be PRing this config, it should be available in the next releases

Mrcubix, 2021-03-25:
just to make sure, go to tablet > tablet debugger, move your pen to the bottom right corner and check the max position value

iceCreamMint, 2021-03-25:
max values are [50800, 31750], but most times it goes to [50790, 31740]

PR #938 (2021-03-30): Added GC610 Configuration
Fixes #916

path issues should be solved
git stay trash, just like vscode github extension
closed 2021-03-30 by InfinityGhost