OpenTabletDriver/OpenTabletDriver #918: Add support for XP-Pen Deco Medium

tcmmichaelb139, 2021-03-25:
# Support Request
There is already "support"(missing features) for the Deco Pro Small 
## Product Information
Manufacturer - XP-Pen
Product Name - Deco Pro Medium
Size - 11" x 6"
Pressure Levels - 8192 levels
LPI - 5080

tcmmichaelb139, 2021-03-26:
Btw, it's just that it doesn't recognize the Deco Pro Medium.

X9VoiD, 2021-03-26:
OTD -> Help -> Export Diagnostics, upload the diagnostics file here.

tcmmichaelb139, 2021-03-26:
[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6212120/diagnostics.txt)

X9VoiD, 2021-03-26:
OTD -> Tablet -> Device string reader -> VID: 10429, PID: 2308 -> Dump all

Upload dump here

tcmmichaelb139, 2021-03-26:
It didn't work. It just kept on showing "replug your tablet, them press OK" I replugged it and it still came up with the same popup.

X9VoiD, 2021-03-26:
Did you properly type in 10429 for VID and 2308 for PID before pressing Dump all?

tcmmichaelb139, 2021-03-26:
Yeah, I did
[image]

X9VoiD, 2021-03-26:
Extract [Deco Pro Medium.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6212305/Deco.Pro.Medium.zip) to `Configurations/XP-Pen/Deco Pro Medium.json` then press OTD -> Tablet -> Detect Tablet.

If you installed OTD through a package manager, the `Configurations` should be in `/usr/share/OpenTabletDriver`

tcmmichaelb139, 2021-03-26:
It works, thanks so much. Only application that I found that could work.

X9VoiD, 2021-03-26:
For verification purposes, open OTD -> Tablet -> Tablet Debugger, then try to get the highest possible value for `Position`

tcmmichaelb139, 2021-03-26:
Its fine. The MaxY is good but the MaxX is 55798 instead of 55780

X9VoiD, 2021-03-26:
Double check if you can't get any higher than 55798

tcmmichaelb139, 2021-03-26:
Yeah I did I got 55798, couldn't get any higher.

tcmmichaelb139, 2021-03-26:
Also I could only get 31399 for MaxY

X9VoiD, 2021-03-26:
To get support for the wheels, open tablet debugger then check which bytes correspond to mechanical wheel, and virtual (touch) wheel, and what those bytes mean if you could.

tcmmichaelb139, 2021-03-26:
To turn right for the mechanical wheel - 02 F0 00 00 00 00 00 01 00 00 
and to turn left it is - 02 F0 00 00 00 00 00 02 00 00 

I couldn't get the virtual wheel because it is just a mouse and I don't know how to get it to work.

X9VoiD, 2021-03-26:
Thanks for the information! Support for the mechanical wheel will be added in the future.

tcmmichaelb139, 2021-03-26:
There is also another issue. Not sure if I should just open another issue but its that when you press down the pen it doesn't register it as any click

X9VoiD, 2021-03-26:
Check if tablet debugger is reading pressure correctly, and if it is, check if you have a pen tip binding set up

tcmmichaelb139, 2021-03-26:
I did, and the it says that there is no pressure.

X9VoiD, 2021-03-26:
Are you able to tell which bytes represent pressure data?

tcmmichaelb139, 2021-03-26:
Yes, the second byte from the left changes from A0 to A1

X9VoiD, 2021-03-26:
The pressure should be contained within two bytes, to represent all 8192 levels of pressure

tcmmichaelb139, 2021-03-26:
Its the only that changes when I put pressure on it. The following 4 bytes after it change based on the position and rest stay the same

X9VoiD, 2021-03-26:
It shouldn't be possible since that means your tablet or pen only supports two levels of pressure, none or max pressure, and nothing in between. Are you sure that you don't see anything else changing when applying different levels of pressure?

On standard tablets the pressure is reported on the 7th and 8th byte, right after the 4 positional bytes (2 for X and 2 for Y)

If you don't see such thing then it's possible that the pressure is reported through a separate interface.

tcmmichaelb139, 2021-03-26:
The 2nd byte is A0 or A1 the 3rd and 4th change based on the height above the tablet and the 4th and 6th change when there is a change on the position on the tablet. All 7, 8, 9, and 10 are 00

tcmmichaelb139, 2021-03-26:
I fixed it. I had to replug my tablet. So sorry.

X9VoiD, 2021-03-26:
All's good :bread:

InfinityGhost, 2021-03-27:
Reopening to link to #924

closed 2021-03-27 by InfinityGhost

okitoxo, 2021-07-21:
@X9VoiD I hope support for mechanical and touch wheels