OpenTabletDriver/OpenTabletDriver #919: My Tablet

RinOkumurra, 2021-03-25:
## Description
I can't Install the driver and if i want to use Osu!lazer my Tablet isnt getting detected.
So i tried to Install the OTD but i click on Install its just Instantly done but it doesnt work, i tried to open it again but it just says "Install" so idk what to do.

## System Information:

| Name    RinOkumurra     
| ---------------- | ----- |
| Operating System |Windows 10
| Software Version |
| Tablet           | Gaomon S620

Mrcubix, 2021-03-25:
[image]

X9VoiD, 2021-03-25:
Choose either HID-compliant device or USB Input Device as the new driver to install

RinOkumurra, 2021-03-25:
https://imgur.com/v8Rhsz9
then
https://imgur.com/brAhSEx
then
https://imgur.com/RPVPeuF
ok but what now i doesnt understand

X9VoiD, 2021-03-25:
[image]

RinOkumurra, 2021-03-25:
ok and then what?

Mrcubix, 2021-03-25:
> Choose either HID-compliant device or USB Input Device as the new driver to install

RinOkumurra, 2021-03-25:
ok got it thanks <3

X9VoiD, 2021-03-25:
If the issue is already resolved you may close this issue now.

closed 2021-03-25 by InfinityGhost