OpenTabletDriver/OpenTabletDriver #921: please add Huion H420 support

Devidovo, 2021-03-25:
Other drivers act strangely on my tablet and this is the lats driver that i need to check for compatability because my tablet wack

Mrcubix, 2021-03-25:
Your tablet will only work on linux since it possess a firmware revision that doesn't allow us to detect it on windows

peppy, 2021-03-29:
@Mrcubix for the record can you explain this further? what part of the driver stops it from being detected? is this something common across all h420s?

X9VoiD, 2021-03-29:
We refer to these tablets as HVAN tablets. (Device strings are filled with HVAN)

Affected tablets are H420, 420, some revisions of New 1060 Plus, and others that I might have forgotten.

All of the exposed endpoints/interfaces for device is exclusively opened by Windows, rendering us unable to access the tablet. The OEM driver gains access to it by installing WinUSB on top of the target endpoint (which prevents Windows from opening the handle to the device exclusively)

Before, we encountered many problems about unaccessible tablets. Due to that, this [PR](https://github.com/InfinityGhost/HIDSharpCore/pull/14) was developed and determines in advance if the device is going to be opened exclusively. It is based on [Microsoft's documentation](https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/top-level-collections-opened-by-windows-for-system-use)

peppy, 2021-03-29:
Seems quite familiar in my recollection of dealing with these. I still have a firmware flasher and some various firmwares for 420s, so I might take a look and see if we can arrive on a better default firmware for use with these.

AbstractQbit, 2021-03-29:
@peppy h420 and a bunch of other older tablets don't have the instruction to turn on vendor defined mode. Without that mode on modern windows versions the hidclass driver opens it exclusively for system, so nothing else can read from it. Vendor drivers get around that by swapping hidclass for winusb filter driver, but OTD doesn't support winusb yet (the lib we use only supports hid).

AbstractQbit, 2021-03-29:
Oops, was typing too slow and didn't see Void's reply

X9VoiD, 2021-03-29:
Oh, uh, I linked the wrong documentation. Here's the [documentation](https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/hid-architecture#hid-clients-supported-in-windows) that was actually used as reference in the `CanOpen` thing, just in case that other people also check the issue.

SaphiraKai, 2021-05-02:
If it has issues on Windows, is there any way I can get it working on Linux at least? I have a Huion 420, and I'd REALLY like to be able to play osu!lazer on it. I'm fine with doing some weird hackery or using an unofficial OpenTabletDriver patch if that will solve the problem.

Mrcubix, 2021-05-02:
tablet work on anything but windows, with a normal config, you can ask for a config from another linux 420 user on OTD's discord https://discord.gg/P3GYdUS

Mrcubix, 2021-09-09:
Join the [discord server](https://discord.gg/P3GYdUS) when you have time, WinUSB support has been added in pre-release and it will be faster to get it compatible if you join

TeamDman, 2021-12-03:
For anyone looking for something in the mean time I downloaded official drivers from [here](https://www.huion.com/index.php?m=content&c=index&a=lists&catid=16&myform=2&down_title1=420(2048)&down_os=Windows&down_type=&category=undefined) that give the ability to resize the input zone and remap buttons

jamesbt365, 2021-12-28:
support added #1426

PR #1426 (2021-12-19): Added H420 Configuration
# Verification

https://discord.com/channels/615607687467761684/885680644003340353/885703150743486494
H420 is just Huion 420 with 3 buttons (the normal 420 has 3 solder pads for buttons on PCB)

# Changes:

- Fixes #921 

# Tablet Owner / Contributor:

@MishterKirby
closed 2021-12-28 by jamesbt365