OpenTabletDriver/OpenTabletDriver #925: Vertical squashing

C0D3-M4513R, 2021-03-26:
## Description

Using my normal Area feels off, like it was vertically squashed. 
Is everything alright in that code section?
This was really clearly the case, on first startup.

I changed my area ratio from 1.7778 to a lower 1.6883 .
This felt more normal.

I then restarted a OTD a couple times, and checked again.
The strange feeling was gone.
I then set my area ratio to 1.4444 . This should be really low, but felt just slightly too little vertical movement. Not a whole lot, and definately still playable.

Is the Vertical part of the Tablet Area working as expected?
Also, while at it, check the Horizontal part too, because I didn't test that.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10 
| Software Version | v.0.5.2.3
| Tablet           | Wacom MTE-450

X9VoiD, 2021-03-26:
Go to OTD -> Tablet -> Tablet Debugger, get the highest possible value for `Position`

C0D3-M4513R, 2021-03-26:
14760, 9225
(Same as current config.)

X9VoiD, 2021-03-26:
Are you entirely sure? Because if yes then this is a non-issue. If it was then all of the current users + osu team should have noticed this "vertical squashing" already.

C0D3-M4513R, 2021-03-26:
[image]
Also VMulti Absoulte Mode or Absoulute Mode doesn't matter.

C0D3-M4513R, 2021-03-26:
> 
> 
> Are you entirely sure? Because if yes then this is a non-issue. If it was then all of the current users + osu team should have noticed this "vertical squashing" already.

It was only really noticable on first start (got used to it quickly, or it went away), but now I can tell, that the Vertical Tablet Area isn't affecting the actual tablet area as much as it should. So maybe this is just due to users accepting some wierdness after an update?

X9VoiD, 2021-03-26:
> Also used v.0.5.2 for a long time, and that felt file.

Nothing changed between v0.5.2 and v0.5.2.3 when it comes to input.

This is not an issue and you're experiencing placebo. Either that, or the reported max is wrong.

C0D3-M4513R, 2021-03-26:
k. I wanted to make sure. Closing then, due to lack of repoducibility.

closed 2021-03-26 by C0D3-M4513R