OpenTabletDriver/OpenTabletDriver #928: Support Request:

EikelFlo, 2021-03-27:
# Support Request
My Gaomon s56k is not getting detected.

## What I've already tried
I tried the Linux FAQ Points:

1. ArgumentOutOfRangeException: Value [0-15]
2. The driver fails to open the tablet (DeviceIOException)
3. Failed to initialize virtual tablet/pointer

I can fix either OutOfRange error but as soon as I replug my tablet I get DeviceIOException and the other way around.
I even tried to use it with installing the digimend-dkms driver beforehand. 

Diagnostics:
https://paste.eikelflo.com/dutilufupa.json

EikelFlo, 2021-03-27:
If there is anything I read wrong my bad

Mrcubix, 2021-03-27:
lemme make a config for your variation real quick

Mrcubix, 2021-03-27:
[S56K.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6216451/S56K.zip)
extract this config in Configurations/Gaomon , replacing the old one and you should be good unless you're using latest master, in which case, let me know

EikelFlo, 2021-03-27:
After crashing the program and checking that there was a small mistake in the json double , It detects it perfectly.
Thanks for the quick help

EikelFlo, 2021-03-28:
After the detection issue was fixed I didn't test out if it does work, and it does not, when I move my stylus on the tablet my mouse doesn't move at all anything I need to enable?. It does get detected in the tablet debugger.

EikelFlo, 2021-03-28:
What I already tried is 

- `sudo nano /etc/modprobe.d/blacklist.confecho "blacklist wacom" | sudo tee -a /etc/modprobe.d/blacklist.conf same with gaomon`
-  sudo chmod 0666 /dev/uinput
-  INPUT_UINPUT=y
- Enabling it with xinput --list and then xinput enable
- gsettings reset org.gnome.desktop.peripherals.touchpad send-events

Mrcubix, 2021-03-28:
let's try some stuff, with the debugger open:
- drag, and tell me the reportID, 
- hover, need it again
- out, YEP, also need that

once you're done with that, try to get max value for position by moving slowly to the bottom right corner, and send the value here too.

EikelFlo, 2021-03-28:
drag : 96 report rate smth with 400
hover:same
out: id 0 report 205

max Pos 32767 , 32767

Mrcubix, 2021-03-28:
as for position, are you sure that's the max?

EikelFlo, 2021-03-28:
[image]

EikelFlo, 2021-03-28:
[image]

Mrcubix, 2021-03-28:
that's weird considering this tablet sensor isn't a square afaik but ok

Mrcubix, 2021-03-28:
[S56K.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6217351/S56K.zip)
try this config

EikelFlo, 2021-03-28:
Still wont move anything works fine in the debugger tho, any touch pad // tablet support I could've enabled or didn't // disabled somewhere?

Mrcubix, 2021-03-28:
[S56K.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6217360/S56K.zip)
going for the pepega clap WR, here is a config where `"StartInclusive": true`

EikelFlo, 2021-03-28:
works fine now :joy:

PR #939 (2021-04-18): Fixed S56K configuration
Fixes #928
closed 2021-04-01 by EikelFlo