OpenTabletDriver/OpenTabletDriver #933: Compatabilty on M1 Macbooks? (Big Sur)

Chanw11, 2021-03-28:
# Support Request
Big Sur/ M1 Macbook doesnt open .app after unziping
Will ask to open from unknown source and will do nothing after

## What I've already tried
Tried versions 0.5.1, 0.5.2, 0.5.2.3
Tried right click open

## Diagnostic Information
none

Mrcubix, 2021-03-28:
M1 is an ARM architecture CPU, it's completely from x86 or x64, you could try building it for your architecture, but we won't support it

Chanw11, 2021-03-28:
> M1 is an ARM architecture CPU, it's completely different from x86 or x64, you could try building it for your architecture, but we don't support it yet

Ok I'll go the emulation route and see how that goes. Thanks!

closed 2021-03-28 by Chanw11