OpenTabletDriver/OpenTabletDriver #936: Veikk A15 Pro Bindings don't override default official bindings

Squaduck, 2021-03-29:
## Description

Bindings don't override default official bindings
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System |Windows 10 Pro (64bit)
| Software Version |v0.5.2.3
| Tablet           |Veikk A15 Pro

X9VoiD, 2021-03-29:
OTD -> Help -> Export Diagnostics, upload file here in zip format

Squaduck, 2021-03-29:
[Diagnostics.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6223692/Diagnostics.zip)

X9VoiD, 2021-03-29:
The tablet buttons for Veikk tablets are not supported yet. If you want you may send us information about which bytes represent which button through tablet debugger to speed up adding support for it.

Squaduck, 2021-03-29:
In Tablet Debugger I didn't see any bytes change when any button was pressed

X9VoiD, 2021-03-29:
Is the "official binding" something configurable on the Veikk's OEM driver?

Squaduck, 2021-03-29:
Yes, you can change what the buttons do in the OEM driver

InfinityGhost, 2021-03-30:
Potentially a secondary endpoint for auxiliary input? Haven't looked at the diagnostics yet.

InfinityGhost, 2021-07-18:
Stale support request, assuming it has been resolved. This can be reopened in the case this was a mistake.

closed 2021-07-18 by InfinityGhost

Squaduck, 2021-07-19:
This has not been resolved

InfinityGhost, 2021-07-19:
Can discussion be taken into Discord? It would help speed up the process.

Squaduck, 2021-07-19:
Shure