OpenTabletDriver/OpenTabletDriver #944: Support Request (Bug?):  XP Pen Star 03 not detected (Ubuntu 20.10 and 20.04)

tinodesigns, 2021-03-31:
# Support Request
I'm unsure if this is actually a bug. However I'm unable to get my XP Pen Star 03 to work. 

The driver installed fine and seems to be really nice but it is not detecting my Star 03 by XP-Pen even though it is looking for it and it should be supported.

Now I'm unsure if the issue is with your driver. On my main PC I still use Ubuntu 18.04 because on 20.04 it is no longer working out of the box (may be already 19.04 and/or 19.10 but I did not try those).

So I found your driver and thought to give it a try on 20.10 but no luck. 

Strange thing is that Ubuntu recognizes that a table is plugged in but it is not working. I only get an extra options panel in the hardware overview of the settings for "wacom graphic tablet" once plued in. But it says no tablet and no stylus recognized. So Ubuntu seems to recognize some tablet hardware is plugged in but not recognizing or giving me access to the tablet and stylus. 

In Ubuntu 18.04 I have setting options here and tablet and Stylus work like a charm. 

OpenTabletDriver seems to behave similair. In the diagnostics of the OpenTabletDriver it seems to recognize something as well but not detect the tablet and stylus. Please have a look there.

## What I've already tried
Different Ubuntu Versions (18.04 seems to be the latest recognising the XP Pen Star 03 without additional drivers)

Also tried the XP-Pen driver for Linux but it did not work and also seems to require Ubuntu to recognize the tablet first. So I removed it again.

Different USB slots.

In OpenTabletDriver app I tried to detect the tablet with no luck. 

## Diagnostic Information
attached

Appreciate your help and driver. I would love to use it, it looks awesome with all the settings you can make. 
Hoepfully my issue can help improve it a bit. Feel free to turn it into a bug if you think it is one.
[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6239024/diagnostics.txt)

X9VoiD, 2021-03-31:
For some reason HidSharpCore did not throw any DeviceIOException when it should have (you don't have access to raw device as indicated by 0 input/output length on all of them)

Perform instructions from [FAQ](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ#the-driver-fails-to-open-the-tablet-deviceioexception) to fix it.

tinodesigns, 2021-03-31:
Hi thanks for the quick reply and input. 

I tried the FAQ option but getting an error when running ./build.sh

`/home/tino/OpenTabletDriver-udev/OpenTabletDriver.udev/OpenTabletDriver.udev.csproj : error NU3028: Paket "System.CommandLine.Experimental 0.3.0-alpha.19577.1" aus Quelle "https://api.nuget.org/v3/index.json": Der Zeitstempel der Primärsignatur des Autors hat ein Kettenerstellungsproblem festgestellt: UntrustedRoot: self signed certificate in certificate chain
/home/tino/OpenTabletDriver-udev/OpenTabletDriver.udev/OpenTabletDriver.udev.csproj : error NU3037: Paket "System.CommandLine.Experimental 0.3.0-alpha.19577.1" aus Quelle "https://api.nuget.org/v3/index.json": Die Gültigkeitsdauer der Primärsignatur des Autors ist abgelaufen.
/home/tino/OpenTabletDriver-udev/OpenTabletDriver.udev/OpenTabletDriver.udev.csproj : error NU3028: Paket "System.CommandLine.Experimental 0.3.0-alpha.19577.1" aus Quelle "https://api.nuget.org/v3/index.json": Der Zeitstempel der Repositorygegensignatur hat ein Kettenerstellungsproblem festgestellt: UntrustedRoot: self signed certificate in certificate chain
/home/tino/OpenTabletDriver-udev/OpenTabletDriver.udev/OpenTabletDriver.udev.csproj : error NU3028: Paket "System.Memory 4.5.3" aus Quelle "https://api.nuget.org/v3/index.json": Der Zeitstempel der Primärsignatur des Autors hat ein Kettenerstellungsproblem festgestellt: UntrustedRoot: self signed certificate in certificate chain
/home/tino/OpenTabletDriver-udev/OpenTabletDriver.udev/OpenTabletDriver.udev.csproj : error NU3037: Paket "System.Memory 4.5.3" aus Quelle "https://api.nuget.org/v3/index.json": Die Gültigkeitsdauer der Primärsignatur des Autors ist abgelaufen.
/home/tino/OpenTabletDriver-udev/OpenTabletDriver.udev/OpenTabletDriver.udev.csproj : error NU3028: Paket "System.Memory 4.5.3" aus Quelle "https://api.nuget.org/v3/index.json": Der Zeitstempel der Repositorygegensignatur hat ein Kettenerstellungsproblem festgestellt: UntrustedRoot: self signed certificate in certificate chain

Fehler beim Buildvorgang. Beheben Sie die Buildfehler, und versuchen Sie es anschließend noch mal.`

German System it basically tells me to fix the building process and try again. And a bunch of wron timestamps of the author signature and chain building issues based on wrong time stamps.

Looks like an invalid SSL certificate to me. (UntrustedRoot: self signed certificate in certificate chain) It is always a wrong timestamp (Zeitstempel). My system's time and date are correct. 

Anything else I can check?

tinodesigns, 2021-04-01:
Thanks a lot. 

So I added the rules and finished running the commands in the FAQ but no luck. Tablet is still not detected. I'm attaching my diagnostics again may be you can see a difference.

Also I noticed the build.sh file defined a configuration file as well which was not there. Was the build.sh script supposed to create it or should it have been there already? I just noticed it when pasting the 99-opentabletdriver.rules to its appropriate folder.

[diagnostics-v2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6242008/diagnostics-v2.txt)

X9VoiD, 2021-04-01:
```
  "HID Devices": [],
```
You're somehow in an even worse situation now.

The commands in the FAQ are confirmed working as that is what we're using to generate a udev rule, so unless you've done something incorrectly you should not be having any problems right now.

tinodesigns, 2021-04-01:
> ```
>   "HID Devices": [],
> ```
> 
> You're somehow in an even worse situation now.
> 
> The commands in the FAQ are confirmed working as that is what we're using to generate a udev rule, so unless you've done something incorrectly you should not be having any problems right now.

I noticed that as well. So I tried again since Ubuntu still seemed to at least recognize some tablet device is plugged in. This time I get the diagnostics attached. It doesn't seem much different to the first one to me though.

Also please see my other post. I'm unsure if the commands ran properly since there is also a configuration file and path in the build.sh that however was missing. Does it have to be in there for build.sh to work or should build.sh create it? neither worked.

Thanks
[diagnotics_v3.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6244379/diagnotics_v3.txt)

X9VoiD, 2021-04-01:
Use [this](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6240445/99-opentabletdriver.rules.zip) rule that Infinity sent you, extract and move it to `/etc/udev/rules/d/99-opentabletdriver.rules` then `sudo udevadm control --reload-rules`, replug tablet then press "Detect tablet" on OTD

tinodesigns, 2021-04-01:
I don't have /etc/udev/rules/d/ only /etc/udev/rules.d/
I also can't create folders there currently. May be with the terminal and sudo?

X9VoiD, 2021-04-01:
Was typo.

tinodesigns, 2021-04-01:
k thanx I'll try it again

tinodesigns, 2021-04-01:
Well still no luck. diagnostics is attached.

Anything else I can try? 

[diagnostics_v4.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6244527/diagnostics_v4.txt)

X9VoiD, 2021-04-01:
[99-opentabletdriver.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6244588/99-opentabletdriver.zip)

Update your `99-opentabletdriver.rules` with this one, then export diagnostics

tinodesigns, 2021-04-01:
OTD ist giving me empty diagnostics and seems to quit during export of diagnostics.
Rebooting and will try again.

tinodesigns, 2021-04-01:
Same issue after reboot. OTD crashes during diagnostics export. Diagnostics file is generated but is empty. I can't attach it.

tinodesigns, 2021-04-01:
BTW thanks for the great response. 
Anyway to support you at least a bit if we can get this resolved?

X9VoiD, 2021-04-01:
See if you can get a log by using the console application of OTD. Run it with root.

tinodesigns, 2021-04-01:
can't run it as root I get this error in the terminal for sudo opentabletdriver and it crashes before it opened

`tino@tino-SATELLITE:~$ sudo opentabletdriver
[sudo] Passwort für tino: 
Failed to connect to bus: Datei oder Verzeichnis nicht gefunden
Failed to connect to bus: Datei oder Verzeichnis nicht gefunden

(OpenTabletDriver.UX.Gtk:3730): libnotify-WARNING **: 17:58:34.862: Failed to connect to proxy
Exception in Gtk# callback delegate
  Note: Applications can use GLib.ExceptionManager.UnhandledException to handle the exception.
System.UnauthorizedAccessException: Could not connect to the pipe because it was not owned by the current user.
   at System.IO.Pipes.NamedPipeClientStream.ValidateRemotePipeUser(SafePipeHandle handle)
   at System.IO.Pipes.NamedPipeClientStream.TryConnect(Int32 timeout, CancellationToken cancellationToken)
   at System.IO.Pipes.NamedPipeClientStream.ConnectInternal(Int32 timeout, CancellationToken cancellationToken, Int32 startTime)
   at System.IO.Pipes.NamedPipeClientStream.<>c__DisplayClass20_0.<ConnectAsync>b__0()
   at System.Threading.Tasks.Task.InnerInvoke()
   at System.Threading.Tasks.Task.<>c.<.cctor>b__277_0(Object obj)
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot, Thread threadPoolThread)
--- End of stack trace from previous location ---
   at OpenTabletDriver.Desktop.RPC.RpcClient`1.Connect()
   at OpenTabletDriver.UX.MainForm.InitializeAsync()
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_0(Object state)
   at Eto.GtkSharp.Forms.GtkSynchronizationContext.<>c__DisplayClass0_0.<Post>b__0()
   at GLib.Idle.IdleProxy.Handler()
   at GLib.ExceptionManager.RaiseUnhandledException(Exception e, Boolean is_terminal)
   at GLib.Idle.IdleProxy.Handler()
   at Eto.GtkSharp.Forms.ApplicationHandler.Run()
   at Eto.Forms.Application.Run(Form mainForm)
   at OpenTabletDriver.UX.App.Run(String platform, String[] args)
   at OpenTabletDriver.UX.Gtk.Program.Main(String[] args)
`

Everyhting I can get from the OTD console is this:

`[Detect:Info]	Searching for tablet 'Wacom PTH-651'
[Detect:Info]	Searching for tablet 'Wacom PTK-540WL'
[Detect:Info]	Searching for tablet 'Wacom CTL-680'
[Detect:Info]	Searching for tablet 'Wacom CTL-671'
[Detect:Info]	Searching for tablet 'Wacom CTL-6100'
[Detect:Info]	Searching for tablet 'Wacom CTH-680'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100WL'
[Detect:Info]	Searching for tablet 'Wacom CTL-471'
[Detect:Info]	Searching for tablet 'Wacom PTH-660 Bluetooth'
[Detect:Info]	Searching for tablet 'Wacom PTH-850'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100'
[Detect:Info]	Searching for tablet 'Wacom CTH-480'
[Detect:Info]	Searching for tablet 'Wacom PTK-440'
[Detect:Info]	Searching for tablet 'Wacom PTZ-930'
[Detect:Info]	Searching for tablet 'Wacom XD-0608-U'
[Detect:Info]	Searching for tablet 'Wacom CTL-480'
[Detect:Info]	Searching for tablet 'Wacom CTH-490'
[Detect:Info]	Searching for tablet 'Wacom PTH-660'
[Detect:Info]	Searching for tablet 'Wacom CTL-470'
[Detect:Info]	Searching for tablet 'Wacom FT-0405-U'
[Detect:Info]	Searching for tablet 'Wacom CTL-672'
[Detect:Info]	Searching for tablet 'Wacom CTH-460'
[Detect:Info]	Searching for tablet 'Wacom PTH-860'
[Detect:Info]	Searching for tablet 'Wacom CTL-460'
[Detect:Info]	Searching for tablet 'Wacom CTL-472'
[Detect:Info]	Searching for tablet 'Wacom PTK-1240'
[Detect:Info]	Searching for tablet 'Wacom CTL-490'
[Detect:Info]	Searching for tablet 'Wacom CTH-670'
[Detect:Info]	Searching for tablet 'Wacom CTH-690'
[Detect:Info]	Searching for tablet 'Wacom PTH-451'
[Detect:Info]	Searching for tablet 'Wacom CTE-440'
[Detect:Info]	Searching for tablet 'Wacom CTE-430'
[Detect:Info]	Searching for tablet 'Wacom CTL-690'
[... 43 lines cut ...]
[Detect:Info]	Searching for tablet 'XP-Pen G640 v2'
[Detect:Info]	Searching for tablet 'XP-Pen G430'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01 v2'
[Detect:Info]	Searching for tablet 'Gaomon M106K'
[Detect:Info]	Searching for tablet 'Gaomon M106K Pro'
[Detect:Info]	Searching for tablet 'Gaomon M10K'
[Detect:Info]	Searching for tablet 'Gaomon S620'
[Detect:Info]	Searching for tablet 'Gaomon 1060 Pro'
[Detect:Info]	Searching for tablet 'Gaomon S56K'
[Detect:Info]	Searching for tablet 'VEIKK A15'
[Detect:Info]	Searching for tablet 'VEIKK A30'
[Detect:Info]	Searching for tablet 'VEIKK A15 Pro'
[Detect:Info]	Searching for tablet 'VEIKK S640'
[Detect:Info]	Searching for tablet 'VEIKK A50'
[Detect:Info]	No tablet found.
[Settings:Info]	Output mode: Absolute Mode
[Settings:Info]	Display area: [1600x900@<800. 450>:0°],
[Settings:Info]	Tablet area: [0x0@<0. 0>:0°],
[Settings:Info]	Clipping: Enabled
[Settings:Info]	Ignoring reports outside area: Disabled
[Settings:Info]	Tip Binding: [Mouse Button Binding: Left]@1%
[Settings:Info]	Pen Bindings: [0, ], [1, ]
[Settings:Info]	Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ], [6, ], [7, ]
[Settings:Info]	Driver is auto-enabled.
[IPC:Debug]	Connected to a client.`

tinodesigns, 2021-04-01:
Opening it normally from terminal using just 'opentabletdriver':

tino@tino-SATELLITE:~$ opentabletdriver

(OpenTabletDriver.UX.Gtk:4678): Gtk-WARNING **: 18:02:18.412: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node arrow owner GtkTreeMenu)

(OpenTabletDriver.UX.Gtk:4678): Gtk-WARNING **: 18:02:18.462: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node arrow owner GtkTreeMenu)

(OpenTabletDriver.UX.Gtk:4678): Gtk-WARNING **: 18:02:18.978: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node arrow owner GtkTreeMenu)

(OpenTabletDriver.UX.Gtk:4678): Gtk-WARNING **: 18:02:18.979: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node arrow owner GtkTreeMenu)

(OpenTabletDriver.UX.Gtk:4678): Gtk-WARNING **: 18:02:19.007: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node arrow owner GtkTreeMenu)

(OpenTabletDriver.UX.Gtk:4678): Gtk-WARNING **: 18:02:19.008: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node arrow owner GtkTreeMenu)

(OpenTabletDriver.UX.Gtk:4678): Gtk-WARNING **: 18:02:19.234: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node arrow owner GtkTreeMenu)

(OpenTabletDriver.UX.Gtk:4678): Gtk-WARNING **: 18:02:19.235: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node arrow owner GtkTreeMenu)

(OpenTabletDriver.UX.Gtk:4678): Gtk-WARNING **: 18:02:19.278: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node arrow owner GtkTreeMenu)

(OpenTabletDriver.UX.Gtk:4678): Gtk-WARNING **: 18:02:19.279: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node arrow owner GtkTreeMenu)

(OpenTabletDriver.UX.Gtk:4678): Gtk-WARNING **: 18:02:19.354: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node arrow owner GtkTreeMenu)

(OpenTabletDriver.UX.Gtk:4678): Gtk-WARNING **: 18:02:19.355: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node arrow owner GtkTreeMenu)

(OpenTabletDriver.UX.Gtk:4678): Gtk-WARNING **: 18:02:19.398: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node arrow owner GtkTreeMenu)

(OpenTabletDriver.UX.Gtk:4678): Gtk-WARNING **: 18:02:19.399: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node arrow owner GtkTreeMenu)

(OpenTabletDriver.UX.Gtk:4678): Gtk-WARNING **: 18:02:20.354: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node arrow owner GtkTreeMenu)

(OpenTabletDriver.UX.Gtk:4678): Gtk-WARNING **: 18:02:20.355: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node arrow owner GtkTreeMenu)

InfinityGhost, 2021-04-01:
> Opening it normally from terminal using just 'opentabletdriver':
>
> tino@tino-SATELLITE:~$ opentabletdriver
>
> (OpenTabletDriver.UX.Gtk:4678): Gtk-WARNING **: 18:02:18.412: Drawing a gadget with [...]

These errors are fine and unrelated to OpenTabletDriver.

tinodesigns, 2021-04-02:
Ok thanks, what are the other errors from the new 99-opentabletdriver.rules ? 
Why can't I export the diagnostics?

Is there anything else I can do?

Is it may be a general issue with the XP Pen Star 03 and Ubuntu 20?
I wonder why it works flawless with Ubuntu 18? 

I was just hoping the OTD might get it to work.

tinodesigns, 2021-04-02:
@X9VoiD is there anything you can read from the info I sent to you from the console and terminal?

Thanks a bunch

X9VoiD, 2021-04-03:
As Infinity mentioned, that error is external to OTD, and we can only wait to fix it.

tinodesigns, 2021-04-04:
Hi there,

so I was able to finally found someone with a similair issue and a solution on this post: 
https://bbs.archlinux.org/viewtopic.php?id=253659

install xf86-input-evdev and create a file /etc/X11/xorg.conf.d/evdev-tablet.conf with these contents:

Section "InputClass"
	Identifier "evdev tablet catchall"
	MatchIsTablet "on"
	MatchDevicePath "/dev/input/event*"
	Driver "evdev"
EndSection

Then restarted Ubuntu. Tablet and pen worked after second reboot (first one showed a messed up screen).

I followed the advice and my stylus and tablet work now. However OTD still is not detecting it. I re-added and re-loaded the rules that worked to export diagnostics and expoerted them. I attached them here. Is there anything else you can do now that the stylus and tablet are working fine?

It would be great to be able to make some settings.

Thanks
[diagnostics_v2.2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6253782/diagnostics_v2.2.txt)

X9VoiD, 2021-04-04:
Export diagnostics, is all what we need. And the stuff you added will only make it worse for you by forcing another driver to load. OTD can't work properly when another driver is running.

X9VoiD, 2021-04-04:
Remove this thing.
```
Section "InputClass"
Identifier "evdev tablet catchall"
MatchIsTablet "on"
MatchDevicePath "/dev/input/event*"
Driver "evdev"
EndSection
```

It'll only conflict with OTD, and the diagnostics still show that you don't have the updated [rule](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6244588/99-opentabletdriver.zip) set up properly. Make sure that you update `/etc/udev/rules.d/99-opentabletdriver.rules` with the new one and then run:
```
sudo udevadm control --reload-rules
sudo udevadm trigger
```

Restart daemon and otd gui then re-export diagnostics.

tinodesigns, 2021-04-06:
I wasn't aware, that this is s differnet driver taht would create issues.So i removed it, and readded those rules. Rebooted and tried again. However I'm still unable to export the diagnostics. 

I did get an export when the tablet was not plugged in. When it is plugged in it crashes during export and the file is empty. It is not yet recognizing the tablet but obviously must be ding something or the export would not crash OTD. 

I attached both diagnostics with and without plugged in tablet. 

I'm not sure if this will help much. If it does not I would need another way of exporting the diagnostics. The console shows the same stuff as before in my above post. 

Thanks a bunch

[diagnostics-tablet-not-plugged-in.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6263081/diagnostics-tablet-not-plugged-in.txt)

X9VoiD, 2021-04-06:
I only see one diagnostics file.

tinodesigns, 2021-04-06:
here is the other sorry. It is empty though since OTD crashes during export when the tablet is plugged in.

X9VoiD, 2021-04-06:
Run OpenTabletDriver through a terminal, then export diagnostics. It should show you what causes the crash.

tinodesigns, 2021-04-06:
Hi what I get is this exception the moment I export the diagnostics. I can't make much sense of it. I just don't know enough what to read out of it. Anything you can see here?  Thanks

tino@tino-SATELLITE:~$ opentabletdriver
Exception in Gtk# callback delegate
  Note: Applications can use GLib.ExceptionManager.UnhandledException to handle the exception.
Newtonsoft.Json.JsonSerializationException: Error getting value from 'MaxInputReportLength' on 'HidSharp.Platform.Linux.LinuxHidDevice'.
 ---> System.ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
   at HidSharp.Reports.Units.Unit.DecodeExponent(UInt32 value)
   at HidSharp.Reports.ReportDescriptor.ParseMainData(MainItemTag tag, UInt32 value)
   at HidSharp.Reports.ReportDescriptor.ParseMain(MainItemTag tag, UInt32 value)
   at HidSharp.Reports.ReportDescriptor.ParseEncodedItem(EncodedItem item)
   at HidSharp.Reports.ReportDescriptor.ParseEncodedItems(IEnumerable`1 items)
   at HidSharp.Reports.ReportDescriptor.ParseRawReportDescriptor(Byte[] buffer)
   at HidSharp.Reports.ReportDescriptor..ctor(Byte[] buffer)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at HidSharp.HidDevice.get_MaxInputReportLength()
   at lambda_method98(Closure , Object )
   at Newtonsoft.Json.Serialization.ExpressionValueProvider.GetValue(Object target)
   --- End of inner exception stack trace ---
   at Newtonsoft.Json.Serialization.ExpressionValueProvider.GetValue(Object target)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalWriter.CalculatePropertyValues(JsonWriter writer, Object value, JsonContainerContract contract, JsonProperty member, JsonProperty property, JsonContract& memberContract, Object& memberValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalWriter.SerializeObject(JsonWriter writer, Object value, JsonObjectContract contract, JsonProperty member, JsonContainerContract collectionContract, JsonProperty containerProperty)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalWriter.SerializeValue(JsonWriter writer, Object value, JsonContract valueContract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerProperty)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalWriter.SerializeList(JsonWriter writer, IEnumerable values, JsonArrayContract contract, JsonProperty member, JsonContainerContract collectionContract, JsonProperty containerProperty)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalWriter.SerializeValue(JsonWriter writer, Object value, JsonContract valueContract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerProperty)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalWriter.SerializeObject(JsonWriter writer, Object value, JsonObjectContract contract, JsonProperty member, JsonContainerContract collectionContract, JsonProperty containerProperty)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalWriter.SerializeValue(JsonWriter writer, Object value, JsonContract valueContract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerProperty)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalWriter.Serialize(JsonWriter jsonWriter, Object value, Type objectType)
   at Newtonsoft.Json.JsonSerializer.SerializeInternal(JsonWriter jsonWriter, Object value, Type objectType)
   at Newtonsoft.Json.JsonSerializer.Serialize(JsonWriter jsonWriter, Object value, Type objectType)
   at Newtonsoft.Json.JsonConvert.SerializeObjectInternal(Object value, Type type, JsonSerializer jsonSerializer)
   at Newtonsoft.Json.JsonConvert.SerializeObject(Object value, Type type, Formatting formatting, JsonSerializerSettings settings)
   at Newtonsoft.Json.JsonConvert.SerializeObject(Object value, Formatting formatting, JsonSerializerSettings settings)
   at Newtonsoft.Json.JsonConvert.SerializeObject(Object value, Formatting formatting)
   at OpenTabletDriver.Desktop.Diagnostics.DiagnosticInfo.ToString()
   at OpenTabletDriver.UX.MainForm.ExportDiagnostics()
   at OpenTabletDriver.UX.MainForm.<ConstructMenu>b__11_14(Object sender, EventArgs e)
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_0(Object state)
   at Eto.GtkSharp.Forms.GtkSynchronizationContext.<>c__DisplayClass0_0.<Post>b__0()
   at GLib.Idle.IdleProxy.Handler()
   at GLib.ExceptionManager.RaiseUnhandledException(Exception e, Boolean is_terminal)
   at GLib.Idle.IdleProxy.Handler()
   at Eto.GtkSharp.Forms.ApplicationHandler.Run()
   at Eto.Forms.Application.Run(Form mainForm)
   at OpenTabletDriver.UX.App.Run(String platform, String[] args)
   at OpenTabletDriver.UX.Gtk.Program.Main(String[] args)

X9VoiD, 2021-04-06:
Follow instructiuons from [FAQ](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ#argumentoutofrangeexception-value-0-15) then export again.

tinodesigns, 2021-04-06:
No sucess yet, diagnostics file is still empty and OTD crashes. I only tried the FAQ that you linked to. Should I try any other?
Here is my terminal input and output:

tino@tino-SATELLITE:~$ echo "blacklist hid_uclogic" | sudo tee -a /etc/modprobe.d/blacklist.conf
[sudo] Passwort für tino: 
blacklist hid_uclogic
tino@tino-SATELLITE:~$ sudo rmmod hid_uclogic
tino@tino-SATELLITE:~$ opentabletdriver
Exception in Gtk# callback delegate
  Note: Applications can use GLib.ExceptionManager.UnhandledException to handle the exception.
Newtonsoft.Json.JsonSerializationException: Error getting value from 'MaxInputReportLength' on 'HidSharp.Platform.Linux.LinuxHidDevice'.
 ---> System.ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
   at HidSharp.Reports.Units.Unit.DecodeExponent(UInt32 value)
   at HidSharp.Reports.ReportDescriptor.ParseMainData(MainItemTag tag, UInt32 value)
   at HidSharp.Reports.ReportDescriptor.ParseMain(MainItemTag tag, UInt32 value)
   at HidSharp.Reports.ReportDescriptor.ParseEncodedItem(EncodedItem item)
   at HidSharp.Reports.ReportDescriptor.ParseEncodedItems(IEnumerable`1 items)
   at HidSharp.Reports.ReportDescriptor.ParseRawReportDescriptor(Byte[] buffer)
   at HidSharp.Reports.ReportDescriptor..ctor(Byte[] buffer)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at HidSharp.HidDevice.get_MaxInputReportLength()
   at lambda_method98(Closure , Object )
   at Newtonsoft.Json.Serialization.ExpressionValueProvider.GetValue(Object target)
   --- End of inner exception stack trace ---
   at Newtonsoft.Json.Serialization.ExpressionValueProvider.GetValue(Object target)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalWriter.CalculatePropertyValues(JsonWriter writer, Object value, JsonContainerContract contract, JsonProperty member, JsonProperty property, JsonContract& memberContract, Object& memberValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalWriter.SerializeObject(JsonWriter writer, Object value, JsonObjectContract contract, JsonProperty member, JsonContainerContract collectionContract, JsonProperty containerProperty)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalWriter.SerializeValue(JsonWriter writer, Object value, JsonContract valueContract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerProperty)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalWriter.SerializeList(JsonWriter writer, IEnumerable values, JsonArrayContract contract, JsonProperty member, JsonContainerContract collectionContract, JsonProperty containerProperty)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalWriter.SerializeValue(JsonWriter writer, Object value, JsonContract valueContract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerProperty)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalWriter.SerializeObject(JsonWriter writer, Object value, JsonObjectContract contract, JsonProperty member, JsonContainerContract collectionContract, JsonProperty containerProperty)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalWriter.SerializeValue(JsonWriter writer, Object value, JsonContract valueContract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerProperty)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalWriter.Serialize(JsonWriter jsonWriter, Object value, Type objectType)
   at Newtonsoft.Json.JsonSerializer.SerializeInternal(JsonWriter jsonWriter, Object value, Type objectType)
   at Newtonsoft.Json.JsonSerializer.Serialize(JsonWriter jsonWriter, Object value, Type objectType)
   at Newtonsoft.Json.JsonConvert.SerializeObjectInternal(Object value, Type type, JsonSerializer jsonSerializer)
   at Newtonsoft.Json.JsonConvert.SerializeObject(Object value, Type type, Formatting formatting, JsonSerializerSettings settings)
   at Newtonsoft.Json.JsonConvert.SerializeObject(Object value, Formatting formatting, JsonSerializerSettings settings)
   at Newtonsoft.Json.JsonConvert.SerializeObject(Object value, Formatting formatting)
   at OpenTabletDriver.Desktop.Diagnostics.DiagnosticInfo.ToString()
   at OpenTabletDriver.UX.MainForm.ExportDiagnostics()
   at OpenTabletDriver.UX.MainForm.<ConstructMenu>b__11_14(Object sender, EventArgs e)
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_0(Object state)
   at Eto.GtkSharp.Forms.GtkSynchronizationContext.<>c__DisplayClass0_0.<Post>b__0()
   at GLib.Idle.IdleProxy.Handler()
   at GLib.ExceptionManager.RaiseUnhandledException(Exception e, Boolean is_terminal)
   at GLib.Idle.IdleProxy.Handler()
   at Eto.GtkSharp.Forms.ApplicationHandler.Run()
   at Eto.Forms.Application.Run(Form mainForm)
   at OpenTabletDriver.UX.App.Run(String platform, String[] args)
   at OpenTabletDriver.UX.Gtk.Program.Main(String[] args)

tinodesigns, 2021-04-06:
Ok after several reboots and trying the FAQs command again I finally was able to export the diagnostics.

It is attached here. Let me know if this is now running wtth the latest rules you sent me and if I need to update anything else.

Thanks
[diagnostics4.1.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6264316/diagnostics4.1.txt)

tinodesigns, 2021-04-06:
Also looks like at least pen pressure is working now.

X9VoiD, 2021-04-06:
Go to OTD -> Tablet -> Device string reader. Then do:
```
VID: 21827
PID: 129
```
then press `Dump`. Upload it here.

tinodesigns, 2021-04-06:
Here you go:
[string-reader-dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6264718/string-reader-dump.txt)

tinodesigns, 2021-04-08:
Did the file help you?

X9VoiD, 2021-04-08:
Yes, the problem is that on the Discord server there's another person of the exact same "model" that reached out to us, and ended up with a very different internal layout for the hardware. So until more information is gathered this issue will stay open for a while.

tinodesigns, 2021-04-08:
Ok. Are you talking about different hardware of the computer the tablet is running on or different hardware of the tablet and pen?

If you need more info on different system hardware I  might be able to get you the data when plugging the tablet into my desktop. 
The info I sent you as from my laptop.

Thanks

tinodesigns, 2021-04-08:
Also would it help to get you some data using my tablet on a Win10 system? I have one running as well and I think there is a windows version of OTD also, right?

X9VoiD, 2021-04-08:
I'm talking about the hardware of the tablet or firmware, whatever it is that was different that made the endpoint layout different. Yours follow Huion protocol while the other follows XP-Pen's protocol.

tinodesigns, 2021-04-08:
Ok thanks. It is infact an XP-PEN I could send you and image with seril and products number etc. but I'm unsure if that actully helps you. 

If you do have an update or need any info, let me know. I'd love to use it. It's only working s plug and play in Ubuntu 18.4.  and I can't figure out what driver is being used there. 

I might just see if I can use that other driver in the mean time. Even though I have no setting options with that at all. 

Besides I'd like to help OTD to be able to support as mny tablets as possible.

tinodesigns, 2021-04-29:
Just FYI.

I'm n contact with XP-Pen. They told me there are actually two versions of the XP-Pen Star 03. The newer Version is sold as XP-Pen Star 03 V2.

Mine is an older one and Linux recognizes it as UC-Logic TABLET 1060N

So you are right about two different hardwares or firmwares for the same tablet. 

I was told to use an older XP Pen driver, but that also does'nt work :|

gonX, 2021-04-29:
Unless I am mistaken, this might have been fixed with #1037

tinodesigns, 2021-04-29:
might have been. Is that fix already in the latest OTD release? I would give it a shot

gonX, 2021-04-29:
As mentioned, you'd have to use a pre-release build. There are sadly no `.deb`'s for pre-releases, so you're a bit on your own. Here's the current latest build built from `master`: https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/793611827

tinodesigns, 2021-04-29:
ok this might be just a bit too complicated for me. I might just wait for the next release to test it unless anyone has exact instructions on how to do it.

Thanks a lot

closed 2021-07-09 by InfinityGhost