OpenTabletDriver/OpenTabletDriver #945: Cannot move display area to the right

gaseet, 2021-04-01:
## Description

I cannot move my display area to the right, only to the left, and even though "Lock to usable area" is enabled, I can move it further to the left, outside my usable area. I have two monitors which are both by AOC, E970Sw and G2460PF (primary). The E970Sw is to the left of my main monitor, however, if I change the arrangement, making it work as if on the right, on my display settings, I can move the area normally.

https://user-images.githubusercontent.com/45747276/113239396-08bd6b00-9281-11eb-9b17-c86383f5af9b.mp4

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System |Windows 10 Home
| Software Version |0.5.2.3
| Tablet           |CTH-480

E-Soda, 2021-04-15:
I have the same issue, but even if the dragging does not work, you can still manually specify where the X and Y coordinates are and position your screen that way.

gaseet, 2021-04-16:
You can also align by right clicking... But it's just annoying.

PR #1035 (2021-04-22): Fix area lock when a display is on a negative coordinate
## Changes

- Push bounds rectangle to (0, 0).

Fixes #945