OpenTabletDriver/OpenTabletDriver #946: Add support for Gaomon M1220

pasi123567, 2021-04-01:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | Gaomon
| Product Name   |M1220
| Width (mm)      |258.4
| Height (mm)     |161.5
| Pressure Levels |8192
| LPI             |5080

## Diagnostic Information
[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6243158/diagnostics.txt)

Mrcubix, 2021-04-01:
i need you to head to tablet > device string reader and type the following:
- 9580
- 109
- 201
then send request and paste the obtained string here

pasi123567, 2021-04-01:
GM001_T201_201030

Mrcubix, 2021-04-01:
[M1220.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6243574/M1220.zip)
extract this config in Configurations/Gaomon then head into the UX, go to tablet > detect tablet.
After that, head to tablet > tablet debugger, and i'll need multiple things from there:
first, move your pen around, you'll see value appear.
on the right side, there is a value called ReportID, i need it's value when dragging and when out of range
After that, i need you to get max values for position, (by positioning your pen to the bottom right side i assume), try to do it slowly as we need precise values

pasi123567, 2021-04-01:
I don't think the tablet is getting detected yet for some reason.

Mrcubix, 2021-04-01:
help > export diagnostic

pasi123567, 2021-04-01:
[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6243685/diagnostics.txt)

Mrcubix, 2021-04-01:
make sure WinUSB is uninstalled in programs and features, afaik, it should had been detected

pasi123567, 2021-04-01:
Did not find that in my programs. Tried restarting and still not detected

Mrcubix, 2021-04-01:
alright, last thing i'm going to take a look at:

head to tablet > device string reader, now click on dump instead of typing the infos, and upload the generated fiels here

pasi123567, 2021-04-01:
It says I should replug the tablet and press ok afterwards what I did but I keep getting the same message

pasi123567, 2021-04-01:
Also windows is detecting and because I can move the cursor with the full area

Mrcubix, 2021-04-01:
use https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/706959832 , it should work there

pasi123567, 2021-04-01:
[image]
Still getting this

pasi123567, 2021-04-01:
Wait it is getting detected with that driver I just downloaded when I imported the file

pasi123567, 2021-04-01:
ReportID = 64, out of Range = 0
Position: [51693, 32307]

Mrcubix, 2021-04-01:
can you get more precise values?

pasi123567, 2021-04-01:
I don't know what you mean exactly? 64 and 0 are the only ones I see and Position wise thats the max I can get 100% certain

Mrcubix, 2021-04-01:
can you get more precise values for pos, by going even slower to that bottom right corner

pasi123567, 2021-04-01:
Wait if I really push it out a few times I can get position [51694, 32308], so one point extra

pasi123567, 2021-04-01:
Thats definitely the max I can get. Don't know what more precise would mean but I don't see commas or something like that

Mrcubix, 2021-04-01:
[M1220.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6243986/M1220.zip)
you should be set with this config, i'll be PRing it and it should be available i nthe next actions builds & releases

pasi123567, 2021-04-01:
Alright, is there any reason though that it doesn't get detected on the official driver download?

Mrcubix, 2021-04-01:
what i sent you is the dev build, it's the official driver, only thing changed is some syntax changes and some reworks, bug fixes and stuff, i used one of these changes in the config, which is why it wasn't working in the release

PR #947 (2021-04-18): Added Gaomon M1220 Configuration
Fixes #946
closed 2021-04-18 by InfinityGhost