OpenTabletDriver/OpenTabletDriver #948: error while installing "The author primary signature's timestamp found a chain building issue: UntrustedRoot: self signed certificate in certificate chain"

NBBZ, 2021-04-02:
## Description

while trying to install from the aur package a lot of red messeges flash on the screen which contain the error:
"The author primary signature's timestamp found a chain building issue: UntrustedRoot: self signed certificate in certificate chain"
and then i get the line:
"Failed to restore /home/nbbz/.cache/paru/clone/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Desktop/OpenTabletDriver.Desktop.csproj (in 1.21 min)."
and others very similar to it, just different paths
## System Information:

| Name             | Value |
| ---------------- | ----- |
| arch linux 5.11.11-arch1-1 |
| aur/opentabletdriver-git v0.5.2.1.r0.g0d9ec6f-2 |
| Veikk S640 |

LavaDesu, 2021-04-02:
Please read the pinned comment on the AUR page.

0Karakurt0, 2021-04-02:
Is there some workaround?

LavaDesu, 2021-04-02:
Once again, please read the pinned comment. There is a link to the bug report with workarounds.

0Karakurt0, 2021-04-02:
> There is a link to the bug report with workarounds.
Can you post here workarounds for this exact repository?
I've never worked with dotnet, I have no idea which packages to verify.

There's also this comment:
Comment by Sergio Tridente (TioDuke) - Saturday, 20 March 2021, 21:15 GMT
Unfortunately, [2] and [3] do not solve the issue for me as it did not for many people.

LavaDesu, 2021-04-02:
Please read [this comment](https://bugs.archlinux.org/task/70095#comment197899) by loqs:
> With ca-certificates-mozilla 3.62-1 installed you should have /etc/ssl/certs/VeriSign_Universal_Root_Certification_Authority.pem
Copy that into the /etc/ca-certificates/trust-source/anchors/ directory then update to ca-certificates-mozilla 3.63-1 which will run update-ca-trust extract.
For more details see https://man.archlinux.org/man/update-ca-trust.8

LavaDesu, 2021-04-09:
Dotnet SDK 5.0.5 was released. The update disables signing on Linux, but it is yet to be pushed on arch repos.  
For an alternative fix, you may install dotnet-sdk-bin from the AUR instead of messing around with certificates.

InfinityGhost, 2021-04-12:
Will close once the official `dotnet-sdk` package on the Arch community repo is updated.

InfinityGhost, 2021-04-16:
Official package has been updated, closing this issue.

closed 2021-04-16 by InfinityGhost