OpenTabletDriver/OpenTabletDriver #949: Open Tablet Drivers not detecting  G640 in linux

osufan11, 2021-04-03:
Its not detecting my tablet for some reason. Can anybody help.
When I press detect tablet it wont show up.

osufan11, 2021-04-03:
i also got this out of the console : 
[Detect:Info]	Searching for tablet 'Huion HC16'
[Detect:Info]	Searching for tablet 'Huion H1060P'
[Detect:Info]	Searching for tablet 'Huion Inspiroy Q11K'
[Detect:Info]	Searching for tablet 'Huion osu!tablet'
[Detect:Info]	Searching for tablet 'Huion Kamvas 16'
[Detect:Info]	Searching for tablet 'Huion Kamvas 13'
[Detect:Info]	Searching for tablet 'Huion HS64'
[Detect:Info]	Searching for tablet 'Huion HS611'
[Detect:Info]	Searching for tablet 'Huion H430P'
[Detect:Info]	Searching for tablet 'Huion WH1409 V2'
[Detect:Info]	Searching for tablet 'Huion H640P'
[Detect:Info]	Searching for tablet 'Huion HS610'
[Detect:Info]	Searching for tablet 'Huion H610 Pro'
[Detect:Info]	Searching for tablet 'Huion Inspiroy Q11K V2'
[Detect:Info]	Searching for tablet 'Huion G10T'
[Detect:Info]	Searching for tablet 'Huion Q620M'
[Detect:Info]	Searching for tablet 'Huion H950P'
[Detect:Info]	Searching for tablet 'Huion 1060 Plus'
[Detect:Info]	Searching for tablet 'Huion GT-133'
[Detect:Info]	Searching for tablet 'Huion H610 Pro V2'
[Detect:Info]	Searching for tablet 'Huion New 1060 Plus'
[Detect:Info]	Searching for tablet 'Huion H1161'
[Detect:Info]	Searching for tablet 'XP-Pen G430'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 03'
[Detect:Info]	Searching for tablet 'XP-Pen G540'
[Detect:Info]	Searching for tablet 'XP-Pen G640 v2'
[Detect:Info]	Searching for tablet 'XP-Pen Star G960S Plus'
[Detect:Info]	Searching for tablet 'XP-Pen G640s'
[Detect:Info]	Searching for tablet 'XP-Pen Deco Mini7'
[Detect:Info]	Searching for tablet 'XP-Pen Star 03'
[Detect:Info]	Searching for tablet 'XP-Pen Star 06C'
[Detect:Info]	Searching for tablet 'XP-Pen G640'
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01 v2'
[Detect:Info]	Searching for tablet 'XP-Pen Star 06'
[Detect:Info]	Searching for tablet 'XP-Pen G430S'
[Detect:Info]	Searching for tablet 'XP-Pen Artist Pro 12'
[Detect:Info]	Searching for tablet 'XP-Pen G960 v2'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 02'
[Detect:Info]	Searching for tablet 'XP-Pen G540 Pro'
[Detect:Info]	Searching for tablet 'XP-Pen Deco Pro Small'
[Detect:Info]	Searching for tablet 'Wacom PTH-860'
[Detect:Info]	Searching for tablet 'Wacom CTL-480'
[Detect:Info]	Searching for tablet 'Wacom CTL-680'
[Detect:Info]	Searching for tablet 'Wacom PTH-850'
[Detect:Info]	Searching for tablet 'Wacom CTH-670'
[Detect:Info]	Searching for tablet 'Wacom PTK-440'
[Detect:Info]	Searching for tablet 'Wacom CTL-6100'
[Detect:Info]	Searching for tablet 'Wacom CTE-460'
[Detect:Info]	Searching for tablet 'Wacom MTE-450'
[Detect:Info]	Searching for tablet 'Wacom PTH-651'
[Detect:Info]	Searching for tablet 'Wacom CTL-472'
[Detect:Info]	Searching for tablet 'Wacom PTZ-930'
[Detect:Info]	Searching for tablet 'Wacom CTL-690'
[Detect:Info]	Searching for tablet 'Wacom CTH-470'
[Detect:Info]	Searching for tablet 'Wacom CTH-690'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100WL'
[Detect:Info]	Searching for tablet 'Wacom PTK-540WL'
[Detect:Info]	Searching for tablet 'Wacom CTH-490'
[Detect:Info]	Searching for tablet 'Wacom CTL-671'
[Detect:Info]	Searching for tablet 'Wacom PTH-660'
[Detect:Info]	Searching for tablet 'Wacom FT-0405-U'
[Detect:Info]	Searching for tablet 'Wacom CTH-680'
[Detect:Info]	Searching for tablet 'Wacom CTL-460'
[Detect:Info]	Searching for tablet 'Wacom CTL-471'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100'
[Detect:Info]	Searching for tablet 'Wacom PTK-1240'
[Detect:Info]	Searching for tablet 'Wacom CTL-490'
[... 99 lines cut ...]
[Detect:Info]	Searching for tablet 'Wacom PTK-1240'
[Detect:Info]	Searching for tablet 'Wacom CTL-490'
[Detect:Info]	Searching for tablet 'Wacom CTH-460'
[Detect:Info]	Searching for tablet 'Wacom CTE-440'
[Detect:Info]	Searching for tablet 'Wacom CTL-672'
[Detect:Info]	Searching for tablet 'Wacom CTE-430'
[Detect:Info]	Searching for tablet 'Wacom CTE-650'
[Detect:Info]	Searching for tablet 'Wacom CTH-480'
[Detect:Info]	Searching for tablet 'Wacom CTL-470'
[Detect:Info]	Searching for tablet 'Wacom PTH-451'
[Detect:Info]	Searching for tablet 'Wacom PTH-660 Bluetooth'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100 Bluetooth'
[Detect:Info]	Searching for tablet 'Wacom XD-0608-U'
[Detect:Info]	Searching for tablet 'Gaomon 1060 Pro'
[Detect:Info]	Searching for tablet 'Gaomon M106K'
[Detect:Info]	Searching for tablet 'Gaomon S620'
[Detect:Info]	Searching for tablet 'Gaomon S56K'
[Detect:Info]	Searching for tablet 'Gaomon M106K Pro'
[Detect:Info]	Searching for tablet 'Gaomon M10K'
[Detect:Info]	Searching for tablet 'VEIKK A15 Pro'
[Detect:Info]	Searching for tablet 'VEIKK A15'
[Detect:Info]	Searching for tablet 'VEIKK A30'
[Detect:Info]	Searching for tablet 'VEIKK S640'
[Detect:Info]	Searching for tablet 'VEIKK A50'
[Detect:Info]	No tablet found.

X9VoiD, 2021-04-03:
Read [FAQ](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ#argumentoutofrangeexception-value-0-15)

coinworld, 2021-04-03:
Please read [Linux FAQ](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ) before submitting an issue

tinodesigns, 2021-04-04:
Is your tablet and pen working already? I have the same issue and my pen and tablet were not even working at all at first.

osufan11, 2021-04-07:
@tinodesigns Not at all

osufan11, 2021-04-07:
@X9VoiD Alright

osufan11, 2021-04-07:
@X9VoiD Its not working it's freezing my console.

X9VoiD, 2021-04-07:
The fix from the FAQ is known to work 100% of the time so unless you've done something weird then it should be working by now. `ArgumentOutOfRangeException` is an error due to `hid_uclogic`'s report descriptor override feature.

closed 2021-04-08 by InfinityGhost

osufan11, 2021-04-08:
@X9VoiD 
[image]
its stuck like this.