OpenTabletDriver/OpenTabletDriver #954: Support Request: DontNet error upon attempt to start v0.5.2.3

Crystal-Gear, 2021-04-04:
# Support Request
 When attempting to start the driver after the install is complete, an error pops up saying "To run this application, you need to install a newer version of .NET Core. Would you like to download it now?"

[image]

[image]

## What I've already tried
 I have been unable to find anything on any search engines after looking for around 15 minutes

## Diagnostic Information

X9VoiD, 2021-04-04:
Click "Yes", download and install NET 5 x64 runtime.

InfinityGhost, 2021-04-06:
You must install the .NET 5 x64 Desktop Runtime.

closed 2021-04-06 by InfinityGhost