OpenTabletDriver/OpenTabletDriver #956: Support Request:

ElBurrito1387, 2021-04-05:
# Support Request

The driver detected my tablet when I first downloaded it, but whenever I opened it for the second time it no longer detects my tablet and I was just going to delete it and get the original driver for my tablet, but whenever I attempt to delete the tablet driver or any of the other files with it I am unable to delete the file regardless of what I do. The updater says that the driver is not installed, as I uninstalled it whenever I was first attempting to delete the driver. But my tablet still works and uses the area I had set before I attempted to delete the driver. Also since the updater says that it is not installed, attempting to install the driver through the updater causes the updater to crash.

## What I've already tried
 I have tried to delete the driver directly through my files, but I am unable to do so as the file is in use. I have also tried deleting the parts of the driver individually and none of them can be deleted.

## Diagnostic Information

Mrcubix, 2021-04-05:
first of all, use portable, i consider the installer as being basically deprecated, also they are both the same thing, second, head to help > export diagnostic

InfinityGhost, 2021-04-06:
This might belong more at https://github.com/OpenTabletDriver/OpenTabletDriver.Installer, however we're no longer maintaining it by the next major update.

closed 2021-07-09 by InfinityGhost