OpenTabletDriver/OpenTabletDriver #959: Crash error on startup (MACOS)

SWAGEDGAR, 2021-04-05:
## Description

When clicked yes the error disappears then reappears. If clicked no the whole driver just closes.
[image]

## System Information:
[image]

| Operating System |
MacOS
| Software Version |
0.5.2.1
| Tablet           |
Wacom CTL 472

X9VoiD, 2021-04-05:
Go to OTD -> Help -> Export Diagnostics, then upload it here in zip format.

SWAGEDGAR, 2021-04-05:
> Go to OTD -> Help -> Export Diagnostics, then upload it here in zip format.

Skip to 1:27 I somehow fixed it by clicking Export Diagnostics. (All the time before that is me trying to click the button)

SWAGEDGAR, 2021-04-05:
But if I close OTD it will say the error again for sure.

InfinityGhost, 2021-04-06:
Check if `~/Library/Application Support/OpenTabletDriver` exists, and if it does try deleting it.

SWAGEDGAR, 2021-04-06:
> Check if `~/Library/Application Support/OpenTabletDriver` exists, and if it does try deleting it.

So I was able to delete it but now it wont open just crash a full crash.

nerelly, 2021-05-01:
Same thing was happening to me. What I did to get it working was delete ~/Library/Application Support/OpenTabletDriver, open activity monitor and force quit opentabletdriver daemon.

InfinityGhost, 2021-05-25:
A workaround appears to exist at least for now, closing this as I've been unable to reproduce.

closed 2021-05-25 by InfinityGhost